libimobiledevice/idevicerestore #27: ERROR: TSS request failed status=94,

genhack, 2015-01-16:
Hello everyone, I just installed idevicerestore, but I have a little problem in practice when sending the command gives me this error: 
Found device in Recovery mode
Identified device as iPhone5,1
Latest firmware is iPhone5,1_8.1.2_12B440_Restore.ipsw
Verifying 'iPhone5,1_8.1.2_12B440_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 8.1.2
Product Build: 12B440 Major: 12
INFO: device serial number is F2MJQEJGDTTN
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 2926820310472
Getting ApNonce in recovery mode... cb fa b0 07 ea 77 0e 59 8a 8f b0 8c a3 8b 0b 0e ac 2d 6b 4f 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
SSL: no alternative certificate subject name matches target host name 'gs.apple.com'
Sending TSS request attempt 1... Request URL set to https://17.171.36.30/TSS/controller?action=2
SSL: certificate subject name 'gs.apple.com' does not match target host name '17.171.36.30'
Sending TSS request attempt 2... Request URL set to https://17.151.36.30/TSS/controller?action=2
SSL: certificate subject name 'gs.apple.com' does not match target host name '17.151.36.30'
Sending TSS request attempt 3... Request URL set to http://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.

ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.
)
Sending TSS request attempt 4... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device
root@genhack-HP-Pavilion-dv6-Notebook-PC:/usr/local/bin# 

How can I fix?? thanks to all in advance

goide, 2015-07-08:
You've found the solution to this problem?

prateek1108, 2015-07-31:
Even I am facing the same error when i am trying to update my iphone firmware .please reply me if you find the solution for this problem....

prateek1108, 2015-08-03:
i am also facing the same issue i.e.

Error: Unable to get component: iBEC
Error: Unable to send iBEC to device
Error: Unable to send iBEC

for iphone6,2_8.4_12H143_Rstore.ipsw

please help me out how i can remove this error and erase and update iphone.

FunkyM, 2015-09-24:
Make sure you have no TSS servers running locally and that your OpenSSL is properly compiled with SSL support.
@prateek1108 Your problem is a different one that should be discussed in a separate ticket, but please add more information.

toni90, 2015-10-23:
i have a same problem .... how to make it with linux kali pllzzz help

toni90, 2015-10-27:
@FunkyM how to compile Opennssl with ssl support

aldypratamak, 2016-04-10:
hi @genhack , I have a same problem with you.did you have resolve it?

closed 2017-06-15 by nikias

pintert3, 2020-11-14:
Was there any conclusive solution to this?

nikias, 2020-11-14:
The error messages is coming from Apple's servers and usually means that they don't sign the given Firmware anymore.

pintert3, 2020-11-14:
It gave me a similar error for iOS 12.4.8 for an old iPad, given that it isn't that old, would that still apply?

pintert3, 2020-11-14:
Thanks btw for the quick response, I really appreciate your efforts and help

pintert3, 2020-11-14:
Allow me to answer myself here. You're right @nikias , looks like Apple updated it as of November 5th. Thanks a lot again. So it's a matter of old IPSW.