libimobiledevice/idevicerestore #29: Freezes on 'Sending iBEC'

khyfry, 2015-01-21:
Hi there,

On Recovery mode the iBSS, APTicket and iBEC refuse to send at all, but in DFU mode this appears:

idevicerestore -d -c -e iPhone4,1_6.1_10B142_Restore.ipsw
Reading data from version.xml
NOTE: using cached version data
opening device 05ac:1227...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag NONC in string Apple Inc.
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string Apple Inc.
Setting to configuration 1
Setting to interface 0:0
opening device 05ac:1227...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag NONC in string Apple Inc.
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string Apple Inc.
Setting to configuration 1
Setting to interface 0:0
Found device in DFU mode
opening device 05ac:1227...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag NONC in string Apple Inc.
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string Apple Inc.
Setting to configuration 1
Setting to interface 0:0
Identified device as iPhone4,1
Extracting Restore.plist from IPSW
Product Version: 6.1
Product Build: 10B142 Major: 10
opening device 05ac:1227...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag NONC in string Apple Inc.
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string Apple Inc.
Setting to configuration 1
Setting to interface 0:0
Device supports Image4: false
Custom firmware requested. Disabled TSS request.
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Using cached filesystem from 'iPhone4,1_6.1_10B142_Restore/038-5307-151.dmg'
opening device 05ac:1227...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag NONC in string Apple Inc.
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string Apple Inc.
Setting to configuration 1
Setting to interface 0:0
Extracting iBSS.n94ap.RELEASE.dfu...
Not personalizing component iBSS...
Writing data to iBSS
Sending iBSS (76260 bytes)...
[==================================================] 100.0%
opening device 05ac:1227...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag NONC in string Apple Inc.
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string Apple Inc.
Setting to configuration 1
Setting to interface 0:0
Nonce: 
Setting to configuration 1
Extracting iBEC.n94ap.RELEASE.dfu...
Not personalizing component iBEC...
Writing data to iBEC
Sending iBEC (276964 bytes)...
[=================================                 ]  65.1%

and freezes about 2/3 of the way into Sending iBEC

is it my doing or will my 4S just not have it?

Thanks

Charyas, 2015-09-29:
I get the same trouble. do you get the more discover?

nikias, 2017-06-15:
Custom firmware with 4s doesn't work.

closed 2017-06-15 by nikias