libimobiledevice/idevicerestore #30: 4s restore failed

cradleOFjheffd, 2015-01-29:
root@Shotdown:~# sudo idevicerestore -e ./Desktop/iPhone4,1_8.1.2_12B440_Restore.ipsw
read_file: File cache/version.xml not found
ERROR: Could not load 'cache/version.xml'.
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
Found device in Recovery mode
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is C39GVBZJDTC0
Identified device as iPhone4,1
Extracting BuildManifest from IPSW
ERROR: zip_name_locate: BuildManifesto.plist
Product Version: 8.1.2
Product Build: 12B440
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 2800588864908
checking for local shsh
no local file shsh/2800588864908-iPhone4,1-8.1.2.shsh
Trying to fetch new SHSH blob
Sending TSS request... received SHSH blobs
Extracting filesystem from IPSW
[==================================================] 100.0%
ERROR: Unable to find iBEC entry in TSS response
iBEC will be signed
Extracting iBEC.n94.RELEASE.dfu
ERROR: Unable to find iBEC entry in TSS response
ERROR: Unable to get SHSH blob for TSS iBEC.n94.RELEASE.dfu entry
ERROR: Unable to get component: iBEC
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC

takumikub, 2015-09-11:
[image]

FunkyM, 2015-09-24:
Which OS are you using?
There is an error here ```ERROR: zip_name_locate: BuildManifesto.plist``` that should not be happening.

closed 2017-06-15 by nikias

ciobanasu, 2018-02-01:
Hi all. When I try to restore my iPhone 4s to iOS 6.1.3 I get an error : Device serial number not found when iBEC is sending. Someone can help me?