libimobiledevice/idevicerestore #31: Unable to connect to device in restore mode

alagbari88, 2015-03-08:
plz help me i have this error how can i fix it ? iphone 5c 
idevicerestore -d -e iPhone5,3_8.1.3_12B466_Restore.ipsw 
* About to connect() to itunes.apple.com port 80 (#0)
*   Trying 104.71.28.237...
* connected
* Connected to itunes.apple.com (104.71.28.237) port 80 (#0)
> GET /check/version HTTP/1.1
User-Agent: InetURL/1.0
Host: itunes.apple.com
Accept: */*

* additional stuff not fine transfer.c:1037: 0 0
* HTTP 1.1 or later with persistent connection, pipelining supported
< HTTP/1.1 302 Moved Temporarily
< Server: AkamaiGHost
< Content-Length: 0
< Location: https://itunes.apple.com/WebObjects/MZStore.woa/wa/com.apple.jingle.appserver.client.MZITunesClientCheck/version
< Date: Mon, 02 Mar 2015 18:31:57 GMT
< Connection: keep-alive
< X-Apple-Partner: origin.0
< 
* Connection #0 to host itunes.apple.com left intact
* Issue another request to this URL: 'https://itunes.apple.com/WebObjects/MZStore.woa/wa/com.apple.jingle.appserver.client.MZITunesClientCheck/version'
* About to connect() to itunes.apple.com port 443 (#1)
*   Trying 104.71.28.237...
* connected
* Connected to itunes.apple.com (104.71.28.237) port 443 (#1)
* successfully set certificate verify locations:
*   CAfile: none
  CApath: /etc/ssl/certs
* SSL connection using AES256-SHA
* Server certificate:
* 	 subject: 1.3.6.1.4.1.311.60.2.1.3=US; 1.3.6.1.4.1.311.60.2.1.2=California; businessCategory=Private Organization; serialNumber=C0806592; C=US; postalCode=95014; ST=California; L=Cupertino; street=1 Infinite Loop; O=Apple Inc.; OU=Internet Services Operations for A
* 	 start date: 2014-04-21 00:00:00 GMT
* 	 expire date: 2016-04-20 23:59:59 GMT
* 	 subjectAltName: itunes.apple.com matched
* 	 issuer: C=US; O=VeriSign, Inc.; OU=VeriSign Trust Network; OU=Terms of use at https://www.verisign.com/rpa (c)06; CN=VeriSign Class 3 Extended Validation SSL SGC CA
* 	 SSL certificate verify ok.
> GET /WebObjects/MZStore.woa/wa/com.apple.jingle.appserver.client.MZITunesClientCheck/version HTTP/1.1
User-Agent: InetURL/1.0
Host: itunes.apple.com
Accept: */*

* additional stuff not fine transfer.c:1037: 0 0
* HTTP 1.1 or later with persistent connection, pipelining supported
< HTTP/1.1 200 OK
< Last-Modified: Mon, 02 Mar 2015 17:54:51 GMT
< strict-transport-security: max-age=31536000
< x-apple-translated-wo-url: /WebObjects/MZStore.woa/wa/com.apple.jingle.appserver.client.MZITunesClientCheck/version
< x-apple-orig-url: https://itunes.apple.com/WebObjects/MZStore.woa/wa/com.apple.jingle.appserver.client.MZITunesClientCheck/version
< x-apple-application-site: ST11
< apple-timing-app: 64 ms
< x-apple-request-store-front: <null>
< x-apple-asset-version: 0
< Content-Type: text/xml; charset=UTF-8
< x-apple-date-generated: Mon, 02 Mar 2015 17:54:51 GMT
< x-apple-application-instance: 2004604
< x-frame-options: SAMEORIGIN
< x-webobjects-loadaverage: 0
< x-apple-jingle-correlation-key: JU57AUQKOQEME4LA2DPOL34CKU
< Server: ATS/4.1.0
< X-Apple-ATS-Cache-Key: /17.173.38.164/80/WebObjects/MZStore.woa/wa/com.apple.jingle.appserver.client.MZITunesClientCheck/version/iDevice/https
< Cache-Control: max-age=1345
< Date: Mon, 02 Mar 2015 18:31:58 GMT
< Transfer-Encoding:  chunked
< Connection: keep-alive
< Connection: Transfer-Encoding
< Vary: X-Apple-Store-Front
< X-Apple-Partner: origin.0
< 
[... 477 lines cut ...]
Sending RestoreRamDisk (15290636 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreDeviceTree' in TSS response
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.n48ap.img3...
DEBUG: tss_response_get_data_by_key: No entry 'ApImg4Ticket' in TSS response
DEBUG: tss_response_get_blob_by_entry: No entry 'RestoreDeviceTree' in TSS response
NOTE: No SHSH blob found for component RestoreDeviceTree
Not personalizing component RestoreDeviceTree...
Writing data to RestoreDeviceTree
Sending RestoreDeviceTree (87884 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.n48...
DEBUG: tss_response_get_data_by_key: No entry 'ApImg4Ticket' in TSS response
DEBUG: tss_response_get_blob_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No SHSH blob found for component RestoreKernelCache
Not personalizing component RestoreKernelCache...
Writing data to RestoreKernelCache
Sending RestoreKernelCache (9260300 bytes)...
About to restore device... 
Waiting for device...

ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
ERROR: Unable to restore device

mcjoeng, 2015-03-09:
I have the same problem。Unable to connect to device in restore mode

alagbari88, 2015-03-10:
hi there mcjoeng , i sent a message to your Email can you Reply to me , thanks

Jookyseacap, 2015-03-25:
I'm having the same issue. I can trigger the device restore using the idevicerestore command, but when the unit reboots into recovery mode it can no longer communicate with the device and eventually times out to this error

ERROR: Unable to connect to device in recovery mode
ERROR: Unable to enter recovery mode
ERROR: Unable to place device into recovery mode from Normal mode

alagbari88, 2015-05-15:
the problem is from usbmuxd just type the usbmuxd on terminal,

cusmire, 2015-08-13:
Hello guys! I'm using libimobile on Windows. My main problem is that anytimes the iphone is in recovery or dfu mode libimobile can't find the iphone anymore. however iTunes or other utility like pangu still recognize the iPhone. Please help me out!

alagbari88, 2015-08-14:
on windows i don't use it , i use it in linux ,

Dante130283, 2015-08-18:
used sudo usbmuxd but result is still the same:
ERROR: Unable to find component node for RestoreKernelCache
ERROR: Unable to get path for component 'RestoreKernelCache'
ERROR: Unable to send RestoreKernelCache to device.
ERROR: Unable to send KernelCache
ERROR: Unable to place device into restore mode

Frustrating...

alagbari88, 2015-08-19:
can you tell me what is your version linux ?
and how install it the tools ?

Dante130283, 2015-08-19:
ubuntu 15.04
everything is installed correctly the problem now is that it stops while it transfers the firmware to the iphone.

by the way, i suggest that you use the following command: idevicerestore -d -e -l

alagbari88, 2015-08-21:
your command is wrong

Dante130283, 2015-08-21:
why do you say it is wrong?

alagbari88, 2015-08-22:
this " idevicerestore -d -e -l" command is wrong . read the command help of
every one what they do and send to me -d -e -l . and i will tell you .

Dante130283, 2015-08-22:
-d debug
-e erase
-l install latest firmware

alagbari88, 2015-08-23:
can you waiting a few days to complete download kali, and re-install the
tools again

Dante130283, 2015-08-26:
Sure i will wait

alagbari88, 2015-08-30:
i completed install new kali and libimobiledevice, i have some question for
you:
are you root ?
how install it libimobiledevice ?
the real problem is with usbmuxd it is not default to connect between
iphone and linux and some lib installed from other .

before get this error :
ERROR: Unable to find component node for RestoreKernelCache
ERROR: Unable to get path for component 'RestoreKernelCache'
ERROR: Unable to send RestoreKernelCache to device.
ERROR: Unable to send KernelCache
ERROR: Unable to place device into restore mode

type in new termnal usbmax . and Make sure you are root .
for restore command if you need to download the last ipws :
 idevicerestore -d  -l

alagbari88, 2015-08-30:
usbmuxd -vv -f output

2015-08-30 8:48 GMT-04:00 Hossain Alagbari <alagbari88@gmail.com>:

arvin15, 2015-09-11:
Hello, I installed usbmxd v 1.1 but still stuck on waiting for device part... Any new development ?

alagbari88, 2015-09-11:
how install the tools from the beginning ؟

DarkUD, 2015-09-12:
Any luck with that issue ? I installed usbmuxd 1.1 but still no luck. I m still blocked at waiting for device

alagbari88, 2015-09-12:
i install it 50 times , to fix all error , and learn how install it step by
step if you add other lib that get error

DarkUD, 2015-09-12:
Then it worked for you ? o_o ?

alagbari88, 2015-09-14:
tools works fine but not baybass iclud

DarkUD, 2015-09-14:
Hmmm I m still stuck to the waiting device phase tho... maybe I should reinstall everything again ... In which order did you install the tools ?

alagbari88, 2015-09-15:
latest tools from here https://github.com/libimobiledevice

DarkUD, 2015-09-15:
Okay . Will try to install from the git.I used the tar.gz provided here : http://www.libimobiledevice.org/

DarkUD, 2015-09-21:
I m getting really really tired ...Tell me, which command did you use in the end for restore ? and which firmware, maybe the key of the problem is there ...

alagbari88, 2015-09-21:
no the key of problem is usbmux , and other lib use it from others source .
tell step pay step how install it ?

FunkyM, 2015-09-24:
This is already way too off-topic.
Please guys use some chatroom/IRC, not this issue system for communication.
@alagbari88
Which OS do you use exactly?
The usbmuxd daemon should be started automatically in the background.
Are you able to run ideviceinfo in all situations?
I think your problem is that you are not using "sudo" to start idevicerestore.

alagbari88, 2015-09-30:
for me i use kali default i am root , i don't need to use sudo , and
usbmuxd not Always
 started automatically in the background.

toni90, 2015-10-27:
@alagbari88 hello how to compile opennssl with ssl support ??

alagbari88, 2015-10-28:
you are missing the lib

iCreamFX, 2016-07-28:
I NEED HELP! I'm using the libmobiledevice on my windows 8.1 and sudo isn't a command in the terminal. 'nor is usbmuxd. I am stuck on this error and there are no solutions. I need my iPhone to have no setup.app so I can access it. Why? because no sim cards work on it. My brother tried fixing it and he is really good at technology but still no luck. Can someone please help?

Dante130283, 2016-07-28:
If you're using Windows 8.1 use Itunes

Tiago Silva

Por favor, antes de imprimir esta mensagem lembre-se que todos somos
responsáveis pela preservação do Ambiente.
Remember that everyone is responsible for the preservation of the
environment so, please think before you print this message.

iCreamFX, 2016-07-29:
@Dante130283 Why do I have to think before I "print" this message. I did absolutely nothing wrong. All I said I need help on this error as well.

Dante130283, 2016-07-29:
@iCreamFX 

That is because i replied directly through my email. Who said you did something wrong?

Sum1n, 2016-08-09:
you can use this video to fix this problem restore idevice on kali linux using libmobiledevice
https://www.youtube.com/watch?v=qRin2ggBGjw

alagbari88, 2016-08-09:
ok thanks i will watch the video

closed 2017-06-15 by nikias

yogendra123456, 2020-05-18:
Hi All,

i use libmobile device library to restore the device using idevicerestore but i got the error here...
device is not entering **normal mode** to **recovery mode**....
i don't getting the solution , while iTunes update the deivce firmwhere. 

***********************************************************************
debug("DEBUG: Waiting for device to connect in recovery mode...\n");
	cond_wait_timeout(&client->device_event_cond, &client->device_event_mutex, 60000);
	cond_wait_timeout(&client->device_event_cond, &client->device_event_mutex, 60000);//add by yogendra 18-5-2020
	if (client->mode != &idevicerestore_modes[MODE_RECOVERY] || (client->flags & FLAG_QUIT)) {
		mutex_unlock(&client->device_event_mutex);
		error("ERROR: Failed to enter recovery mode\n");
		return -1;
	}
*************************************************************************

Thanks in advanced..