libimobiledevice/idevicerestore #32: Unable to discover device mode

mcjoeng, 2015-03-09:
mcjoeng@mcj:~/iPhone Software Updates> idevicerestore -e -c iPhone6,2_8.1.3_12B466_Restore.ipsw
NOTE: using cached version data
ERROR: Unable to discover device mode. Please make sure a device is attached.
mcjoeng@mcj:~/iPhone Software Updates>

FunkyM, 2015-09-24:
Are you able to run ideviceinfo in the first place?

Poulpoutine, 2015-10-18:
I have the same problem, ideviceinfo tells me "No device found, is it plugged in?
" but i can see that there is the recovery icon on my iphone 4 which is obviously plugged in.

THOMASCameroun, 2016-02-26:
i have the same issue ..... ios 9.2.1 iphone5 cdma

Sn0wDev, 2016-05-01:
try running it as root
so:
sudo idevicerestore your.ipsw
it worked for me, after raging for about 15 minutes :P
:package:

18chablis, 2016-09-18:
Unable to discover device mode. Please make sure a device is attached
i am able to run it and i have downloaded every packages for idevicerestore. how can i fix this problem?

samtulach, 2016-11-24:
Thanks, Sn0wDev.
`SUDO` worked for me.

closed 2017-06-15 by nikias

rihorn, 2018-12-06:
I am still getting the error `Unable to discover device mode`

I'm using `idevicerestore - 'x.ipsw'` under the `su` `root`

mounaiban, 2019-02-14:
I'm just going to repeat what Sn0wDev said in https://github.com/libimobiledevice/idevicerestore/issues/32#issuecomment-216081626 in a more SEO-friendly format for all our lost friends across the internets to find us. I was having this same problem when this trick fixed it for me.

If you are having a problem connecting to your iOS device with `idevicerestore`, with a message like

`ERROR: unable to discover device mode, please make sure a device is attached`

try running `idevicerestore` as root using `sudo`, with your device connected and *in either DFU or Restore Mode*:

`sudo -l -u -i 0x00000000BABECAFE`

or if you want to erase your device:

`sudo -l -e -u -i 0x00000000BABECAFE`

As usual, substitute the `-l -e -u` with the right options for what you are doing!
Just in case distros matter, I am currently running Fedora 29 Workstation.

### PROTIP: ECID in dmesg (DFU or Recovery Mode only)
Your device's ECID should show up in an entry in the kernel ring buffer log when you connect it _while it is in DFU or Recovery Mode_. Type `dmesg` after connecting your device and check for messages like:
```
usb-2-2: Product: Apple Mobile Device (DFU Mode)
usb-2-2: Manufacturer: Apple Inc.
usb-2-2: SerialNumber: CPID [omitted...] ECID:00000000BABECAFE [omitted...] 
```
Alternatively just `dmesg | grep -n2 ECID`

Thanks to devnull for the tip, as seen on this ~2013 article *iPhone 3GS – restore or upgrade firmware of iOS devices using Linux* on the Sloppy Linux Notes blog: https://sysadm.mielnet.pl/tag/idevicerestore/

nikias, 2019-02-14:
@mounaiban I fixed a couple of things in your comment. You should not use -u option when you are using -i ECID, it is always best to just use ECID anyway. Also starting from normal mode is also just fine, maybe you have to run it as root, depending on your setup.
However running it as root is only required if libirecovery is not correctly set up via udev. The user running idevicerestore only needs to be member of the group specified in 39-libirecovery.rules and then it shoud work just fine.

nh2, 2019-04-13:
Hey, I came here via Google.

Would it make sense to change the message

```
ERROR: Unable to discover device mode. Please make sure a device is attached.
```

into

```
ERROR: Unable to discover device mode. Please make sure a device is attached and ensure it's in the right mode.
```

? For in my case I had it plugged in but didn't know I had to put it into the recovery mode (or whatever it's called where in off state, you keep the button pressed and insert the cable) first; I thought it can detect it already when it's simply on.

nh2, 2019-04-13:
BTW, thanks for this software, I just successfully reset an old iPad with it.

I also packaged it for nixpkgs/NixOS, so that it can easily be obtained on any Linux distribution: https://github.com/NixOS/nixpkgs/pull/59427

nikias, 2019-04-18:
@nh2 no it doesn't make much sense to change that because it can detect it in all modes. If it fails to detect a device in normal mode then this is usually because usbmuxd isn't properly set up on the system or not running for some reason.

nh2, 2019-04-18:
@nikias Ah I see. Indeed `usbmuxd` was not running for me, as I just used `idevicerestore` directly. Is it correct that for that usage, the device needs to be in the recovery mode?

nikias, 2019-04-21:
@nh2 well that works obviously. Recovery or DFU mode. Devices in normal mode will be switched to Recovery by idevicerestore.

mounaiban, 2024-05-17:
Just an update to my [previous comment](https://github.com/libimobiledevice/idevicerestore/issues/32#issuecomment-463598796), incorporating advice from @nikias.

To those coming here from Google (or seeking advice via ChatGPT):

The `Unable to discover device mode. Please make sure a device is attached` error occurs when `idevicerestore` is unable to transmit to the device via USB. This is done on Linux (and most other *nix OSes) by writing to USB node files. By default the node files are not writable by a non-superuser.

While running `idevicerestore` using `sudo` or as a superuser overcomes this problem, it is safer to add yourself a group authorised to write to the devices' USB node files.

To check which group to add your account to, run `cat /lib/udev/rules.d/39-libirecovery.rules`, and look in the `GROUP=` fields. At time of writing, on Fedora 40, the group is `disk`.

To add your account to the `disk` group, run: `sudo usermod -aG disk $(whoami) `

Remember to log out and back in for the membership to take effect. 

Now, just run `idevicerecovery` without sudo, it should just work! :smiley: 

Try reconnecting the device if you are still having trouble.

## Checking permissions
You can verify that you have write access to node files by first looking up the bus address on `lsusb`:
```
...
Bus 001 Device 009: ID 05ac:0069 Apple, Inc. Apple Mobile Device [Recovery Mode]
...
```

In this case, our device will be in `/dev/bus/usb/001/009`. Check the write permissions with `ls -l /dev/bus/usb/001/009`:
```
crw-rw-r--. 1 root disk 189, 11 May 17 13:05 /dev/bus/usb/001/009
```
This is correct, as both the superuser `root` and the group `disk` has read-write permissions, and can transmit to the USB device represented by this node, while any other non-superuser is not allowed to do so.