libimobiledevice/idevicerestore #36: restore iphone6 and iphone6(plus) failed

RonnieLLL, 2015-04-29:
Recently, I restore iPhone7,2_8.3_12F70.ipsw on iphone6 and iphone6(plus) by using idevicerestore,
But Its failed, the error code:ERROR: Unable to send baseband data. is there any solution, thanks for any answer, many thanks!

RonnieLLL, 2015-04-29:
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... Protocol https not supported or disabled in libcurl
Request URL set to https://17.171.36.30/TSS/controller?action=2
Sending TSS request attempt 2... Protocol https not supported or disabled in libcurl
Request URL set to https://17.151.36.30/TSS/controller?action=2
Sending TSS request attempt 3... Protocol https not supported or disabled in libcurl
Request URL set to http://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 4... Empty reply from server
Request URL set to http://17.171.36.30/TSS/controller?action=2
Sending TSS request attempt 5... TSS server returned: STATUS=126&MESSAGE=An internal error occurred

nikias, 2015-09-23:
The libcurl on your system is not built with https support.

FunkyM, 2015-09-24:
Apart from what Nikias said, fixed in git master head.

closed 2015-09-24 by FunkyM