libimobiledevice/idevicerestore #38: Iphone 4s, restore IOS 8.3

goide, 2015-07-08:
*****@***:~# sudo idevicerestore -e '/root/Desktop/iPhone4,1_8.3_12F70_Restore.ipsw' 
read_file: File cache/version.xml not found
ERROR: Could not load 'cache/version.xml'.
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
Found device in Recovery mode
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is C8WHN8R7DT9Y
Identified device as iPhone4,1
Extracting BuildManifest from IPSW
ERROR: zip_name_locate: BuildManifesto.plist
Product Version: 8.3
Product Build: 12F70
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 2439408448904
checking for local shsh
no local file shsh/2439408448904-iPhone4,1-8.3.shsh
Trying to fetch new SHSH blob
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94)
Sending TSS request... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

Which means that error status=94? or error TSS?

aburgh, 2015-07-08:
8.3 is no longer being signed by Apple. Try downloading 8.4 and installing that.

goide, 2015-07-10:
root@kali:~# sudo idevicerestore -e '/root/Desktop/iPhone4,1_8.4_12H143_Restore.ipsw'  
read_file: File cache/version.xml not found
ERROR: Could not load 'cache/version.xml'.
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
Found device in Recovery mode
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is XXXXXXXXXXXXX
Identified device as iPhone4,1
Extracting BuildManifest from IPSW
ERROR: zip_name_locate: BuildManifesto.plist
Product Version: 8.4
Product Build: 12H143
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 2439408448904
checking for local shsh
no local file shsh/2439408448904-iPhone4,1-8.4.shsh
Trying to fetch new SHSH blob
Sending TSS request... received SHSH blobs
Extracting filesystem from IPSW
[==================================================] 100.0%
ERROR: Unable to find iBEC entry in TSS response
iBEC will be signed
Extracting iBEC.n94.RELEASE.dfu
ERROR: Unable to find iBEC entry in TSS response
ERROR: Unable to get SHSH blob for TSS iBEC.n94.RELEASE.dfu entry
ERROR: Unable to get component: iBEC
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC

goide, 2015-07-10:
Now I get this error , you believe it ?

prateek1108, 2015-08-03:
i am also facing the same issue i.e.

Error: Unable to get component: iBEC
Error: Unable to send iBEC to device
Error: Unable to send iBEC

for iphone6,2_8.4_12H143_Rstore.ipsw

please help me out how i can remove this error and erase and update iphone.

FunkyM, 2015-09-24:
Which operating system are you using and how did you compile idevicerestore?

giorgosvas, 2015-09-28:
hi.please i am trying to restore firmware on an iphone 5 and i have this problem:
root@kali:~/idevicerestore# idevicerestore -e iPhone5,2_7.1.2_11D257_Restore.ipsw
read_file: File cache/version.xml not found
ERROR: Could not load 'cache/version.xml'.
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
Found device in Recovery mode
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is F2NJQ3P9DTWH
ERROR: Unable to discover device type
 please can you help me i dont know what to do
thanks in advance

Gaby4, 2015-12-16:
The same problem with iphone 5C(5,3) ,iPhone5,3_9.0.2_13A452_Restore.ipsw

Any update with the issue?

TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94)
Sending TSS request... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

closed 2017-06-15 by nikias