libimobiledevice/idevicerestore #40: Random choice when restoring with multiple units

neilbrownlee, 2015-07-30:
Hi,

If you tell idevicerestore to restore a specific device using --udid or -u and the deviceid returned from idevice_id - l, it starts off the process, downloads (if needed) the new .ipsw but then puts a different unit into restore mode!!

Has anyone experienced this or worked around it? I have a few devices plugged in and was a bit shocked to discover I'd wiped the wrong one :( 

I'll begin running through the code to see if I can find anything obvious but I was just wondering if anyone had noticed and patched themselves.

neilbrownlee, 2015-07-30:
*Update*

It is essentially ignoring the -u or -udid option and picking the first device in normal mode - I'd say this was problem!

FunkyM, 2015-09-24:
I think this has to do with the download codepath. Usually using ECID is the only way to target correctly.

nikias, 2015-09-24:
Found an old patch. Not sure why I never committed it. Should be fixed with commit df665a7bbf891ab69c92aedca2a2d7e85eca7165 now.

FunkyM, 2015-09-25:
Ah great, yep that should fix it. Fixed in git master head. :)

closed 2015-09-25 by FunkyM