libimobiledevice/idevicerestore #41: ASR STATUS: fail  ERROR: Unable to restore device

Nadir3, 2015-08-19:
**command**
idevicerestore.exe -u d9accf101050836ec9d058cc6545d98223ea1ca9 C:\Users\user\iPhone5,1_8.4_12H143_Restore_CFW.ipsw

**Output**
>NOTE: using cached version data
Found device in Recovery mode
Identified device as iPhone5,1
Extracting BuildManifest from IPSW
Product Version: 8.4
Product Build: 12H143 Major: 12
INFO: device serial number is F2LL4DE9FH19
Device supports Image4: false
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Found ECID 1942811966965
Getting ApNonce in recovery mode... fb c1 4c 73 af 11 78 4d 73 e9 d8 d8 d5 ca a1 f0 55 27 07 2c
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode...
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting filesystem from IPSW
Sending APTicket (2738 bytes)
Extracting iBEC.n41.RELEASE.dfu...
Not personalizing component iBEC...
Sending iBEC (287116 bytes)...
Getting ApNonce in recovery mode... fb c1 4c 73 af 11 78 4d 73 e9 d8 d8 d5 ca a1 f0 55 27 07 2c
Sending APTicket (2738 bytes)
Recovery Mode Environment:
iBoot build-version=iBoot-2261.30.37
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~iphone.s5l8950x.img3...
Not personalizing component AppleLogo...
Sending AppleLogo (6796 bytes)...
Extracting 058-15893-139.dmg...
Not personalizing component RestoreRamDisk...
Sending RestoreRamDisk (17330444 bytes)...
Extracting DeviceTree.n41ap.img3...
Not personalizing component RestoreDeviceTree...
Sending RestoreDeviceTree (88524 bytes)...
Extracting kernelcache.release.n41...
Not personalizing component RestoreKernelCache...
Sending RestoreKernelCache (9414284 bytes)...
About to restore device...
Waiting for device...
Device is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 13
Device has successfully entered restore mode
Hardware Information:
BoardID: 0
ChipID: 35152
UniqueChipID: 1942811966965
ProductionMode: true
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Partition NAND device (28)
Waiting for NAND (29)
Waiting for NAND (29)
Checking filesystems (16)
Checking filesystems (16)
Unknown operation (51)
Waiting for NAND (29)
Waiting for NAND (29)
About to send filesystem...
Connected to ASR
Validating the filesystem
[... 374 lines cut ...]
entering ramrod_probe_media
device partitioning scheme is GPT
find_filesystem_partitions: storage=/dev/disk0s1 system=/dev/disk0s1s1 data=/dev/disk0s1s2 baseband data= log=
entering unmount_filesystems
entering restore_images
Successfully marked device node /dev/rdisk0s1s1 as static data
executing /usr/sbin/asr -source asr://localhost:12345 -target /dev/disk0s1s1 -erase -noprompt --chunkchecksum --puppetstrings
SYSLOG: Aug 19 17:29:39 localhost kernel[0] <Notice>: void AppleUSBDeviceMux::handleConnectResult(struct BulkUSBMuxSession *, errno_t) new session to port 12345 failed: 61
SYSLOG: Aug 19 17:29:39 localhost asr[30] <Error>: bootstrap_look_up2 failed with 0x44e
ASR: asr: Unable to disable idle sleep - -536870199
ASR STATUS: start       242.6   multicast-client
ASR: Waiting for connection attempt from server
SYSLOG: Aug 19 17:29:39 localhost asr[30] <Error>: __net_helper_get_connection_block_invoke_3 could not connect to networkd
SYSLOG: Aug 19 17:29:41 localhost kernel[0] <Notice>: AppleUSBDeviceMux::handleConnectResult new session 0x95ee35d8 established 12345<-lo0->49156 12345<-usb->45312
ASR STATUS: setup
ASR: Validating target...
ASR: done
ASR STATUS: metadata
ASR: Validating source...
SYSLOG: Aug 19 17:29:42 localhost kernel[0] <Notice>: void AppleUSBDeviceMux::handleMuxTCPInput(mbuf_t) received reset, closing 0x95ee35d8
ASR: Can't gather image metadata
ASR: Could not validate source - error 110
ASR STATUS: fail

ERROR: Unable to restore device

Itbnolehurr, 2015-09-08:
Here: https://www.youtube.com/user/leonteiz/videos

FunkyM, 2015-09-24:
You can not restore a custom firmware unless you have an iBoot level exploit which was only possible on older devices (limera1n etc.).
```
ERROR: Unable to read OOB data from filesystem offset: No error
ERROR: ASR was unable to validate the filesystem
ERROR: Unable to send filesystem
ERROR: Unable to successfully restore device
```
Attempting to fiddle around with the filesystem images or the ipsw will only make it fail.
Thus closing as not supported with this combination of device and iOS version.

closed 2015-09-24 by FunkyM