libimobiledevice/idevicerestore #42: RestoreKernelCache - iPad 3

MrChenaho, 2015-08-25:
Trying to restore a iPad 3, all seems to go well up until the Sending RestoreDeviceTree, the error occurs just as the device says to connect to iTunes? What could be causing this issues? LOG:

E:\libi>idevicerestore.exe -u --9b1d7d69cea1741db97248450b9e65e896875536 -c  iPad3,1_7.1.2_11D257_Restore.ipsw -e
NOTE: using cached version data
Found device in Recovery mode
Identified device as iPad3,1
Extracting Restore.plist from IPSW
Product Version: 7.1.2
Product Build: 11D257 Major: 11
INFO: device serial number is DYTJ65QQDVD1
Device supports Image4: false
Custom firmware requested. Disabled TSS request.
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Using cached filesystem from 'iPad3,1_7.1.2_11D257_Restore/058-4505-009.dmg'
Extracting iBEC.j1ap.RELEASE.dfu...
Not personalizing component iBEC...
Sending iBEC (291212 bytes)...
Getting ApNonce in recovery mode... f9 8e 5a c2 3e 48 b3 4e 02 22 6a 19 65 76 2f cc 16 26 e2 57
Recovery Mode Environment:
iBoot build-version=iBoot-2261.30.37
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~ipad.s5l8945x.img3...
Not personalizing component AppleLogo...
Sending AppleLogo (17868 bytes)...
Extracting 058-4289-009.dmg...
Not personalizing component RestoreRamDisk...
Sending RestoreRamDisk (8892684 bytes)...
Extracting DeviceTree.j1ap.img3...
Not personalizing component RestoreDeviceTree...
Sending RestoreDeviceTree (81804 bytes)...
ERROR: Unable to find component node for RestoreKernelCache
ERROR: Unable to get path for component 'RestoreKernelCache'
ERROR: Unable to send RestoreKernelCache to device.
ERROR: Unable to send KernelCache
ERROR: Unable to place device into restore mode

nikias, 2015-08-25:
iOS 7.1.2 is not signed by Apple anymore and there is currently no known bootrom exploit that would allow you to restore a custom firmware on your hardware.

closed 2015-08-25 by nikias

iMorphey, 2017-09-19:
Please. Help me to solve this problem:
NOTE: using cached version data
Found device in Recovery mode
Identified device as iPad2,4
Extracting Restore.plist from IPSW
Product Version: 6.1.3
Product Build: 10B329 Major: 10
INFO: device serial number is DMPJ6LJ4DFHW
Device supports Image4: false
Custom firmware requested. Disabled TSS request.
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Using cached filesystem from 'iPad2,4_6.1.3_10B329_Restore/048-2467-005.dmg'
Extracting iBEC.k93aap.RELEASE.dfu...
Not personalizing component iBEC...
Sending iBEC (276964 bytes)...
Getting ApNonce in recovery mode... b2 01 38 f5 dd 49 20 63 f1 99 ba 1f 03 6e e8 22 21 77 2c 1d
Recovery Mode Environment:
iBoot build-version=iBoot-2817.60.2
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo.s5l8942x.img3...
Not personalizing component AppleLogo...
Sending AppleLogo (7524 bytes)...
Extracting 048-2558-005.dmg...
Not personalizing component RestoreRamDisk...
Sending RestoreRamDisk (9951588 bytes)...
Extracting DeviceTree.k93aap.img3...
Not personalizing component RestoreDeviceTree...
Sending RestoreDeviceTree (73636 bytes)...
ERROR: Unable to find component node for RestoreKernelCache
ERROR: Unable to get path for component 'RestoreKernelCache'
ERROR: Unable to send RestoreKernelCache to device.
ERROR: Unable to send KernelCache
ERROR: Unable to place device into restore mode