libimobiledevice/idevicerestore #43: Unable to place device into recovery mode from DFU mode

animus31, 2015-08-26:
Hello 

Please I am blocked , I am not able to restore my Ipod 

ERROR: Unable to connect to recovery device
ERROR: Unable to place device into recovery mode from DFU mode

What can I do ? 

Thanks 

Log . 
C:\Users\LARBI\Desktop\libi>idevicerestore.exe -u --udid iPod5,1_8.4.1_12H321_Restore.ipsw
NOTE: using cached version data
Found device in DFU mode
Identified device as iPod5,1
Extracting BuildManifest from IPSW
Product Version: 8.4.1
Product Build: 12H321 Major: 12
Device supports Image4: false
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Found ECID 2354078594378
Getting ApNonce in dfu mode... b7 0e b7 36 e0 3a 48 7d
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode...
WARNING: Unable to find BbChipID node
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Using cached filesystem from 'iPod5,1_8.4.1_12H321_Restore/058-24051-023.dmg'
Extracting iBSS.n78.RELEASE.dfu...
Personalizing IMG3 component iBSS...
reconstructed size: 80410
Sending iBSS (80410 bytes)...
Nonce: be 5f 6a 27 47 79 e6 dc c0 15 9b 80 3f 23 e4 41 23 aa 72 49
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode...
WARNING: Unable to find BbChipID node
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.n78.RELEASE.dfu...
Not personalizing component iBEC...
Sending iBEC (283020 bytes)...
ERROR: Unable to connect to recovery device
ERROR: Unable to place device into recovery mode from DFU mode

xiaojiangDraug, 2015-09-22:
I also get the trouble. my iphone get the same logs. getlasterror return 997 is send buffer data wrong?do you fix this problem?

FunkyM, 2015-09-24:
If you don't know how to handle this you shouldn't be using this experimental software, especially on unsupported Windows. Nevertheless the issue here is most likely within libirecovery.
Which versions of dependencies did you use when compiling idevicerestore?

GorkemSelcuk, 2015-12-01:
i have got the same problem. anybody got fix?

daniel-bluesea, 2016-06-30:
hi FunkyM,

is there a way to restore a device in dfu mode? 
and on which os? (linux, windows, mac)
is it supported at all?

thanks!

nikias, 2016-06-30:
@daniel-bluesea it definitely works on linux and mac, I tested the latest build myself. And it should work on windows too even though we don't officially support it, as @FunkyM said already.

daniel-bluesea, 2016-07-03:
@nikias it doesnt work for iPhone < 6 (i.e. 5c, se, and similar) even on linux & mac.
i think there is an open ticket for it.

iPhone > 6 (6, 6s, 6s+) works fine even on windows

nikias, 2016-07-04:
@daniel-bluesea I successfully restored an iPod Touch 5th gen from DFU with the latest code on linux and mac.

closed 2017-06-15 by nikias

daniel-bluesea, 2017-06-19:
@nikias cool, thanks! i think this issue was solved sometimes last year. my bad for not updating it on time.

pintert3, 2020-11-14:
While I think my issue could be with the device itself, I want to be sure so I'll ask. Does anyone have any actually solution to this without having to update. Cause I'm already using the latest version, but the device(it's an iPad) only responds to the computer in DFU mode, and yet when it tries to initiate the restore, it seems to take it to restore mode, where it cannot be read. (At least from what I've observed, since it's never read in recovery mode). I thought DFU mode had some kind of safe pass to force the recovery but now I'm stuck, and I don't know if there's actually a software solution to this. Anyone with an opinion on this? Any help?

pintert3, 2020-11-14:
For the record, it seems very related to ticket #324 , and I'm slowly starting to believe that there's an error point when the apple device cannot restart properly in restore mode, and so all recovery software on the desktop/laptop cannot get a new response from the apple device. I don't know if it can be solved from here, but I thought you guys might have an idea. Hopefully it helps with #324 too