libimobiledevice/idevicerestore #44: change the RestoreOptions, restore request also same

xiaojiangDraug, 2015-08-27:


nikias, 2015-08-27:
You can't. The restore options are ignored by restored (the daemon running on the device that controls the actual restore process) and it is using its own hardcoded set of restore options.

xiaojiangDraug, 2015-08-28:
thank you for your answer!

DarkUD, 2015-09-21:
Can you think of a reason why a device can not go into restore mode from recovery mode ?

xiaojiangDraug, 2015-09-21:
I think the phone maybe in lockdownd, the recovery command requests will be refused. Do you get the error logs？

DarkUD, 2015-09-21:
I only get waiting for device and it is stuck in recovery mode since I received it. I tho that might be a device problem. What does it means it is in lockdown ? and how to fix that ?

FunkyM, 2015-09-24:
Please don't abuse the ticket title for your question!
Please don't discuss or ask off-topic questions rather create your own ticket for it.
On topic, as noted by Nikias, you can't set the options on production firmwares.
They only work on Debug and Factory ones.

closed 2015-09-24 by FunkyM