libimobiledevice/idevicerestore #45: idevicerestore fails with iPhone 6 Plus on Windows

jalcock0, 2015-08-28:
I have rebuilt libimobiledevice on windows by using mingw32.  I am attempting to restore a iPhone 6 Plus which is failing when restoring baseband data.

I am using the latest Lib-zip version.  The IPSW opens correctly and begins to flash until it reaches the baseband data.

Does anyone has any insight here?

See logs:

Found device in Normal mode
Identified device as iPhone7,1
Extracting BuildManifest from IPSW
Product Version: 8.4.1
Product Build: 12H321 Major: 12
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID ################
Getting ApNonce in normal mode... 58 7c 55 1a e8 4b 9d 43 96 89 d2 16 e3 f2 6e d1 89 e7 b7 a7 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 6e 64 2d d4 9e df af f5 5e 41 d8 01 66 c8 f2 c2 e6 4c c4 1d 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting filesystem from IPSW
Entering recovery mode...
INFO: device serial number is ############
Extracting iBEC.n56.RELEASE.im4p...
Sending iBEC (648713 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-2261.30.37
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@3x~iphone.t7000.im4p...
Sending AppleLogo (18859 bytes)...
ramdisk-size=0x4000000
Extracting 058-23799-023.dmg...
Sending RestoreRamDisk (17495549 bytes)...
Extracting DeviceTree.n56ap.im4p...
Sending RestoreDeviceTree (168375 bytes)...
Extracting kernelcache.release.n56...
Sending RestoreKernelCache (9462180 bytes)...
About to restore device... 
Waiting for device...
Device is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 13
Device has successfully entered restore mode
Hardware Information:
BoardID: 4
ChipID: 28672
UniqueChipID: ##############
ProductionMode: true
FDRSupport indicated, starting FDR listener thread
About to send NORData...
Found firmware path Firmware/all_flash/all_flash.n56ap.production
Getting firmware manifest Firmware/all_flash/all_flash.n56ap.production/manifest
Extracting LLB.n56.RELEASE.im4p...
Extracting DeviceTree.n56ap.im4p...
Extracting applelogo@3x~iphone.t7000.im4p...
Extracting batterycharging0@3x~iphone.t7000.im4p...
Extracting batterycharging1@3x~iphone.t7000.im4p...
Extracting batteryfull@3x~iphone.t7000.im4p...
Extracting batterylow0@3x~iphone.t7000.im4p...
Extracting batterylow1@3x~iphone.t7000.im4p...
Extracting glyphplugin@1920~iphone-lightning.t7000.im4p...
Extracting iBoot.n56.RELEASE.im4p...
Extracting recoverymode@1920~iphone-lightning.t7000.im4p...
Extracting sep-firmware.n56.RELEASE.im4p...
Extracting sep-firmware.n56.RELEASE.im4p...
Sending NORData now...
Done sending NORData
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Partition NAND device (28)
About to send FDR Trust data...
Sending FDR Trust data now...
Done sending FDR Trust Data
Waiting for NAND (29)
Waiting for NAND (29)
Waiting for NAND (29)
Waiting for storage device (11)
Creating partition map (12)
Creating partition map (12)
Creating partition map (12)
Checking filesystems (16)
Checking filesystems (16)
Checking filesystems (16)
Unknown operation (51)
Waiting for NAND (29)
Waiting for NAND (29)
Waiting for NAND (29)
About to send filesystem...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
Done sending filesystem
Restoring image (14)
Checking filesystems (16)
Checking filesystems (16)
Checking filesystems (16)
About to send KernelCache...
Extracting kernelcache.release.n56...
Sending KernelCache now...
Done sending KernelCache
Loading kernelcache (27)
Unknown operation (18)
Unknown operation (46)
Unknown operation (46)
Flashing NOR (19)
About to send BasebandData...
Sending BasebandData now...
Done sending BasebandData
Flashing NOR (19)
Flashing NOR (19)
Flashing NOR (19)
Flashing NOR (19)
Flashing NOR (19)

Updating Baseband in progress...

28/08/2015 17:19:39 Error of DeviceRestore : WARNING: Unable to find BbSkeyId node
ERROR: could not close and write modified archive: Renaming temporary file failed: No such file or directory
ERROR: device didn't accept BasebandData
ERROR: Unable to successfully restore device
ERROR: Unable to restore device

FunkyM, 2015-08-28:
Yes, simply iPhone 6 and iPhone 6 Plus are not supported. A pull request is pending and will be merged soon to fix it.

bullionmotik, 2016-02-17:
Are you using windows, if yes how did you get libzip form the zip folder. Please I'm stucked there

julian2706, 2016-03-04:
hola me podrian ayudar yo tengo mi iphone 5s y vi este mensaje y el mio quedo fue BUENO DIGAMOS QUE ASI SOLO Q 6,1 BLA,BLA DESCARGUE EL DEL IPHONE 5S GSM Q ES EL MIO 
Dispositivo que se encuentra en el modo Normal 
dispositivo identificado como iPhone7,1 
Extracción de BuildManifest IPSW 
Versión del producto: 8.4.1 
Build producto: 12H321 Mayor: 12 
Dispositivo apoya Image4: verdadera 
variación de borrado cliente Instalar (IPSW) 
Este restaurar borrará los datos del dispositivo. 
Encontrado ECID ################ 
Conseguir ApNonce en modo normal ... 58 55 7c 1a e8 4b 9d 43 96 89 16 e3 d2 d1 f2 6e 89 e7 b7 a7 
al buscar el nuevo SHSH blob 
Conseguir SepNonce en modo normal ... 6e 64 2d d4 f5 9e df af 5e 41 d8 01 66 c8 f2 c2 e6 4c c4 1d 
URL de solicitud establecido en https://gs.apple.com/TSS/controller?action=2
Envío de solicitud de intento de TSS 1 ... respuesta con éxito recibidas 
recibidas SHSH 
Extracción del sistema de archivos IPSW 
de entrar en modo de recuperación ... 
INFO: número de serie del dispositivo es ############ 
Extracción iBEC.n56.RELEASE.im4p ... 
Envío de iBEC (648713 bytes) ... 
el modo de recuperación para el Medio Ambiente: 
iBoot construir-version = iBoot-2261.30.37 
iBoot construir-style = LIBERACIÓN 
Enviando AppleLogo ... 
La extracción de applelogo@3x~iphone.t7000.im4p ... 
Envío de AppleLogo (18859 bytes) ... 
disco RAM-size = 0x4000000 
La extracción 058-23799-023.dmg ... 
enviando RestoreRamDisk (17495549 bytes) ... 
La extracción de DeviceTree.n56ap.im4p ... 
enviando RestoreDeviceTree (168375 bytes). .. 
La extracción kernelcache.release.n56 ... 
Envío de RestoreKernelCache (9462180 bytes) ... 
Acerca de restaurar el dispositivo ... 
Esperando dispositivo ... 
El dispositivo está ahora conectado en modo de restauración ... 
Conexión de ahora ... 
conectado a com.apple.mobile.restored, la versión 13 
del dispositivo ha entrado con éxito en restaurar el modo de 
información de hardware: 
BoardID: 4 
ChipID: 28672 
UniqueChipID: ############## 
ProductionMode: verdadera 
FDRSupport indicado, a partir FDR oyente hilo 
a punto de enviar NORData ... 
ruta del firmware encontrado firmware / all_flash / all_flash.n56ap.production 
Obtención de firmware manifiesta firmware / all_flash / all_flash.n56ap.production / manifiesta 
que extrae LLB.n56.RELEASE.im4p ... 
La extracción de DeviceTree.n56ap. im4p ... 
La extracción de applelogo@3x~iphone.t7000.im4p ... 
La extracción de batterycharging0@3x~iphone.t7000.im4p ... 
La extracción de batterycharging1@3x~iphone.t7000.im4p ... 
La extracción de batteryfull @ 3x ~ iPhone. t7000.im4p ... 
La extracción de batterylow0@3x~iphone.t7000.im4p ... 
La extracción de batterylow1@3x~iphone.t7000.im4p ... 
La extracción de glyphplugin@1920~iphone-lightning.t7000.im4p ... 
La extracción de iBoot. n56.RELEASE.im4p ... 
La extracción de recoverymode@1920~iphone-lightning.t7000.im4p ... 
La extracción Sep-firmware.n56.RELEASE.im4p ... 
La extracción Sep-firmware.n56.RELEASE.im4p ... 
Enviando NORData ahora ... 
realiza enviando NORData 
punto de enviar RootTicket ... 
envío de RootTicket ahora ... 
Hecho enviar RootTicket 
dispositivo de reparto NAND (28) 
QUEDA AY NO SE, MIRO EL IPHONE Y APARECE LA MANZANA CON BARRA PERO NI CARGA, Y CUANDO LO DESCONECTO APARECE LA MANZANA UNOS 20 O MAS SEGUNDOS Y PANTALLA AZUL Y SE DE NUEVO Y LO RESTAURO CON ITUNES PERO PUES QUIERO RESTAURARLO DEL TODO, QUICIERA SABER CUANTO TIEMPO SE DEMORA PARA QUE AVANZE DEL (28) POR FAVOR DIGANME AL MENOS CUANTO ESPERO GRACIAS POR FAVOR ESCRIBIRME A ESTE GMAIL juliangon.2000@gmail.com

nikias, 2017-06-15:
This should work by now. Closing.

closed 2017-06-15 by nikias