libimobiledevice/idevicerestore #46: I need help to intall idevicerestore on Ubuntu

TheCosp, 2015-08-31:
Hello everybody, 

I've downloaded the package from this link: https://github.com/libimobiledevice/idevicerestore
I've extracted the package on Desktop
I've opened the terminal and from the directory idevicerestore-master I've typed the following command:

~/Desktop/idevicerestore-master$ sudo ./autogen.sh
libtoolize: putting auxiliary files in `.'.
libtoolize: linking file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: linking file `m4/libtool.m4'
libtoolize: linking file `m4/ltoptions.m4'
libtoolize: linking file `m4/ltsugar.m4'
libtoolize: linking file `m4/ltversion.m4'
libtoolize: linking file `m4/lt~obsolete.m4'
libtoolize: Consider adding `-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
configure.ac:15: installing './compile'
checking whether make supports nested variables... yes
checking build system type... i686-pc-linux-gnu
checking host system type... i686-pc-linux-gnu
checking target system type... i686-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
/home/thecospirator/Desktop/idevicerestore-master/missing: Unknown `--is-lightweight' option
Try `/home/thecospirator/Desktop/idevicerestore-master/missing --help' for more information
configure: WARNING: 'missing' script is too old or missing
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert i686-pc-linux-gnu file names to i686-pc-linux-gnu format... func_convert_file_noop
checking how to convert i686-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
[... 115 lines cut ...]
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libirecovery... no
configure: error: Package requirements (libirecovery >= 0.2.0) were not met:

Requested 'libirecovery >= 0.2.0' but version of libirecovery is 0.1.1

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libirecovery_CFLAGS
and libirecovery_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
thecospirator@TheCospUbuntu:~/Desktop/idevicerestore-master$ make sudo make install
make: *** No rule to make target `sudo'.  Stop.
thecospirator@TheCospUbuntu:~/Desktop/idevicerestore-master$ sudo make
make: *** No targets specified and no makefile found.  Stop.
thecospirator@TheCospUbuntu:~/Desktop/idevicerestore-master$ sudo make install
make: *** No rule to make target `install'.  Stop.

Where am I wrong?

Jamesmallon1, 2015-09-15:
You must download the latest version of libirecovery

Jamesmallon1, 2015-09-15:
https://github.com/Chronic-Dev/libirecovery

FunkyM, 2015-09-24:
@Jamesmallon1 While it is correct, the linked version from Chronic-Dev is completely outdated.
You have to use libirecovery from this github organsiation.
@TheCosp if you read carefully you'll notice that you are missing a dependency:
```Requested 'libirecovery >= 0.2.0' but version of libirecovery is 0.1.1```
Compile and install that first before compiling idevicerestore.

closed 2015-09-24 by FunkyM

Sum1n, 2016-08-16:
you must install require dependency. it may seem that you need to install pkg-config, clang and gcc to avoit this error in additional you may visit the official lbimobiledevice github repository and use this video to fixed some error
https://www.youtube.com/watch?v=qRin2ggBGjw