libimobiledevice/idevicerestore #48: I need help to complile idevice restore on ubuntu 15.04

Zflowrec, 2015-08-31:
i'm new to linux i installed ubuntu 15.04 on virtualbox and libimobile 2.0 and all th required packages for idevicerestore and i get this error

malddingo@malddingo-VirtualBox:~/idevicerestore$ make
make  all-recursive
make[1]: Entering directory '/home/malddingo/idevicerestore'
Making all in src
make[2]: Entering directory '/home/malddingo/idevicerestore/src'
  CC       idevicerestore-dfu.o
dfu.c: In function ‘dfu_get_cpid’:
dfu.c:243:48: warning: initialization makes pointer from integer without a cast
  const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
                                                ^
dfu.c:248:21: error: dereferencing pointer to incomplete type
  *cpid = device_info->cpid;
                     ^
dfu.c: In function ‘dfu_get_ecid’:
dfu.c:260:48: warning: initialization makes pointer from integer without a cast
  const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
                                                ^
dfu.c:265:21: error: dereferencing pointer to incomplete type
  *ecid = device_info->ecid;
                     ^
dfu.c: In function ‘dfu_is_image4_supported’:
dfu.c:278:48: warning: initialization makes pointer from integer without a cast
  const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
                                                ^
dfu.c:283:21: error: dereferencing pointer to incomplete type
  return (device_info->ibfl & IBOOT_FLAG_IMAGE4_AWARE);
                     ^
dfu.c: In function ‘dfu_get_ap_nonce’:
dfu.c:293:48: warning: initialization makes pointer from integer without a cast
  const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
                                                ^
dfu.c:298:17: error: dereferencing pointer to incomplete type
  if (device_info->ap_nonce && device_info->ap_nonce_size > 0) {
                 ^
dfu.c:298:42: error: dereferencing pointer to incomplete type
  if (device_info->ap_nonce && device_info->ap_nonce_size > 0) {
                                          ^
dfu.c:299:46: error: dereferencing pointer to incomplete type
   *nonce = (unsigned char*)malloc(device_info->ap_nonce_size);
                                              ^
dfu.c:303:28: error: dereferencing pointer to incomplete type
   *nonce_size = device_info->ap_nonce_size;
                            ^
dfu.c:304:29: error: dereferencing pointer to incomplete type
   memcpy(*nonce, device_info->ap_nonce, *nonce_size);
                             ^
dfu.c: In function ‘dfu_get_sep_nonce’:
dfu.c:317:48: warning: initialization makes pointer from integer without a cast
  const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
                                                ^
dfu.c:322:17: error: dereferencing pointer to incomplete type
  if (device_info->sep_nonce && device_info->sep_nonce_size > 0) {
                 ^
dfu.c:322:43: error: dereferencing pointer to incomplete type
  if (device_info->sep_nonce && device_info->sep_nonce_size > 0) {
                                           ^
dfu.c:323:46: error: dereferencing pointer to incomplete type
   *nonce = (unsigned char*)malloc(device_info->sep_nonce_size);
                                              ^
dfu.c:327:28: error: dereferencing pointer to incomplete type
   *nonce_size = device_info->sep_nonce_size;
                            ^
dfu.c:328:29: error: dereferencing pointer to incomplete type
   memcpy(*nonce, device_info->sep_nonce, *nonce_size);
                             ^
Makefile:602: recipe for target 'idevicerestore-dfu.o' failed
make[2]: *** [idevicerestore-dfu.o] Error 1
make[2]: Leaving directory '/home/malddingo/idevicerestore/src'
Makefile:408: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/malddingo/idevicerestore'
Makefile:339: recipe for target 'all' failed
make: *** [all] Error 2

jaggions, 2015-09-15:
I have the exact same problem. Waiting for a solution.

Jamesmallon1, 2015-09-15:
Exact problem here aswell

Jamesmallon1, 2015-09-15:
Tried on Lubuntu, Ubuntu 15.04, Ubuntu 14.04 and ubuntu 12.04.
Installed all dependencies. Will keep trying and keep you all updated

sethasaurus, 2015-09-18:
I can't believe I just went through all the rigmarole of downloading & compiling dependencies of dependencies of dependencies, getting to the point of typing 'make', and the software that SOMEONE ELSE WROTE IS BROKEN!
F*********CK!

sethasaurus, 2015-09-18:
Hours and hours and hours of time to get to a dead end!

Jamesmallon1, 2015-09-18:
I know mate its a joke it hasn't been updated for years..

Jamesmallon1, 2015-09-18:
If anyone has a solotion please share

DarkUD, 2015-09-19:
Lol well you installed it using the git repositories or the ubuntu ones ? I had the same issues but I fixed it by installing everything using the git ones.

Jamesmallon1, 2015-09-19:
Do you have a working compiled version atm?

DarkUD, 2015-09-19:
Yep on ubuntu 14.04

sethasaurus, 2015-09-20:
James - Just a thought. Try doing make clean for everything and then recompiling.

DarkUD, 2015-09-21:
Did you manage to compile it ?

matteyeux, 2015-10-28:
Guys you Elrhk's script here : https://github.com/elrhk/Library-iPhone

Sum1n, 2016-08-16:
if you get this error that mean your compiler package have not properly install or configure
you missed some major dependence like gcc or clang so first of all make sure you have install   Development Packages of:
	libimobiledevice libimobiledevice-dbg libimobiledevice-utils libimobiledevice-dev
	libimobiledevice6-dbg libimobiledevice6 libimobiledevice-doc python-imobiledevice
        libirecovery
	libusbmuxd
	libplist
	libzip
	openssl

Software:
	usbmuxd
	make
	autoheader
	automake
	autoconf
	libtool
	pkg-config
	gcc
and use this video to fixed some problem
https://www.youtube.com/watch?v=qRin2ggBGjw

nikias, 2017-06-15:
Compiles and installs fine since ubuntu 10.04. have been doing this for years (now on 17.04). Standard linux build and install procedure for all related packages.

closed 2017-06-15 by nikias