libimobiledevice/idevicerestore #51: restore device to ios 9.0 failed

RonnieLLL, 2015-09-17:
today I use libimobiledevice to restore my device(iPad air1) to iOS 9.0 failed, Is it have not be surpported?  the question is
"
 if (connection->ssl_data)
the ssl_data is NULL;
"

jalcock0, 2015-09-17:
Same issue shown here iPad4,2 Cellular.

Went to restore mode, progress bar shown, no progression.  Halts on partition NAND data.

jalcock0, 2015-09-18:
I am attempting to restore a iPad Air which is failing when restoring.

Does anyone has any insight here please?

See Logs:

Reading data from version.xml
found device with ECID 00000XXXXXXXXXXXXXXX
Setting to interface 0:0
WARNING: set interface failed, error -8
found device with ECID 00000XXXXXXXXXXXXXXX
Setting to interface 0:0
WARNING: set interface failed, error -8
found device with ECID 00000XXXXXXXXXXXXXXX
Setting to interface 0:0
WARNING: set interface failed, error -8
Getting device's ECID for TSS request
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbFDRSecurityKeyHash node
WARNING: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RecoveryMode
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreDeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreKernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreRamDisk
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreSEP
DEBUG: Adding action ESEC to TSS entry
DEBUG: Adding action EPRO to TSS entry
DEBUG: Applying restore request rules for entry SEP
DEBUG: Adding action ESEC to TSS entry
DEBUG: Adding action EPRO to TSS entry
DEBUG: Applying restore request rules for entry iBEC
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry iBSS
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry iBoot
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: tss_response_get_path_by_entry: No entry 'iBEC' in TSS response
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Writing data to iBEC
found device with ECID 00000XXXXXXXXXXXXXXX
Setting to interface 0:0
WARNING: set interface failed, error -8
DEBUG: tss_response_get_path_by_entry: No entry 'AppleLogo' in TSS response
NOTE: No path for component AppleLogo in TSS, will fetch from build_identity
Writing data to AppleLogo
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreRamDisk' in TSS response
NOTE: No path for component RestoreRamDisk in TSS, will fetch from build_identity
Writing data to RestoreRamDisk
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreDeviceTree' in TSS response
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Writing data to RestoreDeviceTree
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Writing data to RestoreKernelCache
Attempt 1 to connect to restore mode device...
Attempt 2 to connect to restore mode device...
Attempt 3 to connect to restore mode device...
Attempt 4 to connect to restore mode device...
Attempt 5 to connect to restore mode device...
Attempt 6 to connect to restore mode device...
restore_is_current_device: Connected to com.apple.mobile.restored, version 13
Attempt 7 to connect to restore mode device...
DEBUG: tss_response_get_path_by_entry: No entry 'LLB' in TSS response
NOTE: Could not get LLB path from TSS data, will fetch from build identity
Writing data to LLB
Writing data to iBoot
Writing data to DeviceTree
Writing data to AppleLogo
Writing data to RecoveryMode
Writing data to BatteryLow0
Writing data to BatteryLow1
Writing data to BatteryCharging0
Writing data to BatteryCharging1
Writing data to BatteryPlugin
Writing data to BatteryFull
Writing data to RestoreSEP
Writing data to SEP
No data to read
No data to read
No data to read
No data to read

mpenttil, 2015-09-23:
At least for iPhone 5S, it seems that bringing FDRTrustData available allows restore to proceed.

jalcock0, 2015-09-23:
Same problem noted on iPad Air (1/2) and iPad Mini 3.

RonnieLLL, 2015-09-24:
I try it sucessed on iPhone 5,  but 6 and 6pluse have not try yet.  now it seems not allows on ipad only!

mpenttil, 2015-09-24:
iPad mini 3 is presumably different story. For me it has been failing since 8.4.0 went out. Device fails to enter recovery mode.

Now that I have the pull request https://github.com/libimobiledevice/idevicerestore/pull/39/files locally merged, and idevicerestore patched so that it will create FDRThread even if build manifest tells otherwise (tss-request to Apple constructed without ApProductionMode and ApSecurityMode fields) I can restore (iOS 9.0.1) pretty much all the devices I can get my hands on i.e.:
iPads 1, 2, 3 and air2 
iPhone 5S and 6

Hth, I'm running on 64 bit Linux (Ubuntu 14.04).

jalcock0, 2015-09-24:
Please see further logs for failure on iPad Air. Does anyone have any insight on this? We have the up to date branch all merged locally and HTTPS support on Libcurl.

About to send RootTicket...
Sending RootTicket now...
13:37:22 property_list_service.c:127 internal_plist_send(): sending 7084 bytes
13:37:22 service.c:140 service_send(): sending 4 bytes
13:37:22 service.c:140 service_send(): sending 7084 bytes
13:37:22 property_list_service.c:132 internal_plist_send(): sent 7084 bytes
13:37:22 property_list_service.c:133 internal_plist_send(): printing 7084 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>RootTicketData</key>
	<data>
	MIITexYESU00TQIBADGCCZD/hOqFnEKCCYcwggmDFgRNQU5CMYIJef+E6oWcUIIBADCB
	/RYETUFOUDGB9P+EkrmGSB4wHBYEQk5DSAQUfW676JVvXbwf5wm+HTACs2jVl0D/hJK9
	pEQLMAkWBEJPUkQCARL/hJqVoE8LMAkWBENFUE8CAQH/hJqhklANMAsWBENISVACAwCJ
	YP+EmsGkTwswCRYEQ1BSTwEB//+Ems2KQwswCRYEQ1NFQwEB//+Eqo2SRBAwDhYERUNJ
	RAIGB/QqSgAw/4WakZ5NCzAJFgRTRE9NAgEB/4ebud5uHjAcFgRzbm9uBBTu14IiDapw
	r31oUOOsr+anLuHFy/+Hm8nsbh4wHBYEc3J2bgQURirdhN9+ARe0XGsIMiFXFtZzHnf/
	hpOF6DBlMGMWBGJhdDAxW/+Eop2mVB4wHBYEREdTVAQUwvbTshN5gi+YU4NwbiHYV2Qd
	Xab/hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//hKrNikMLMAkWBEVT
	RUMBAf//hpOF6DFlMGMWBGJhdDExW/+Eop2mVB4wHBYEREdTVAQUBrOba/LPmCdOpdcs
	9pbWazL6cx//hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//hKrNikML
	MAkWBEVTRUMBAf//hpOF6EZlMGMWBGJhdEYxW/+Eop2mVB4wHBYEREdTVAQUPjqEloSy
	8AB8yzhLmXTXYd+bKb3/hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//
	hKrNikMLMAkWBEVTRUMBAf//hpuhzjBlMGMWBGNoZzAxW/+Eop2mVB4wHBYEREdTVAQU
	O6dB+ngoLIyHV50S8K+IQ1ZtvYL/hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQ
	Uk8BAf//hKrNikMLMAkWBEVTRUMBAf//hpuhzjFlMGMWBGNoZzExW/+Eop2mVB4wHBYE
	REdTVAQUFL++ptTtim6WXisUxK/5jNwpMZv/hKqtilkLMAkWBEVLRVkBAf//hKrBpE8L
	MAkWBEVQUk8BAf//hKrNikMLMAkWBEVTRUMBAf//hqPR5GVlMGMWBGR0cmUxW/+Eop2m
	VB4wHBYEREdTVAQUdqS4lrx7DBgUImj2xRppJrneJbH/hKqtilkLMAkWBEVLRVkBAf//
	hKrBpE8LMAkWBEVQUk8BAf//hKrNikMLMAkWBEVTRUMBAf//hrux8lBlMGMWBGdseVAx
	W/+Eop2mVB4wHBYEREdTVAQUbGDjznRtKY6OH7W3lTIyuQKCiYT/hKqtilkLMAkWBEVL
	RVkBAf//hKrBpE8LMAkWBEVQUk8BAf//hKrNikMLMAkWBEVTRUMBAf//hsuJymNlMGMW
	BGliZWMxW/+Eop2mVB4wHBYEREdTVAQUb8QXh0psvUb5jK3yUs/NnZSKBxn/hKqtilkL
	MAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//hKrNikMLMAkWBEVTRUMBAf//hsuJ
	3nRlMGMWBGlib3QxW/+Eop2mVB4wHBYEREdTVAQUtdDAUueQQn4Ls105jY6HR79Cwy//
	hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//hKrNikMLMAkWBEVTRUMB
	Af//hsuJ5nNlMGMWBGlic3MxW/+Eop2mVB4wHBYEREdTVAQUWCZxSIcOQiYyXh6vwtpj
	a8pOxvr/hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//hKrNikMLMAkW
	BEVTRUMBAf//hsux2GJlMGMWBGlsbGIxW/+Eop2mVB4wHBYEREdTVAQUThAcu+ktPXJG
	2BK+IttDquk+/kb/hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//hKrN
	ikMLMAkWBEVTRUMBAf//htvJ3GxlMGMWBGtybmwxW/+Eop2mVB4wHBYEREdTVAQUAzHt
	nvZu1cw8l/H89O/wMM+Hzkz/hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8B
	Af//hKrNikMLMAkWBEVTRUMBAf//huO9zm9lMGMWBGxvZ28xW/+Eop2mVB4wHBYEREdT
	VAQU3d0fhiFjqG3E3ugOAXMxT35uDk3/hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkW
	BEVQUk8BAf//hKrNikMLMAkWBEVTRUMBAf//h5OR5mtlMGMWBHJkc2sxW/+Eop2mVB4w
	HBYEREdTVAQUjqDyzt9o3rAzf7GxpxMOUZURsZn/hKqtilkLMAkWBEVLRVkBAf//hKrB
	pE8LMAkWBEVQUk8BAf//hKrNikMLMAkWBEVTRUMBAf//h5OR6HJlMGMWBHJkdHIxW/+E
	op2mVB4wHBYEREdTVAQU9P4hgnA8eTThZOXrfZpjb5H6Qdr/hKqtilkLMAkWBEVLRVkB
	Af//hKrBpE8LMAkWBEVQUk8BAf//hKrNikMLMAkWBEVTRUMBAf//h5OVxm1lMGMWBHJl
	Y20xW/+Eop2mVB4wHBYEREdTVAQUpg2nLhqj6RXZPw6B4Spm0Fi7N1f/hKqtilkLMAkW
	BEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//hKrNikMLMAkWBEVTRUMBAf//h5Ot5G5l
	MGMWBHJrcm4xW/+Eop2mVB4wHBYEREdTVAQU7PRasrsbxujkvXF8e/+7mxkqvf3/hKqt
	ilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//hKrNikMLMAkWBEVTRUMBAf//
	h5Oxzm9lMGMWBHJsZ28xW/+Eop2mVB4wHBYEREdTVAQUxC8NPIS3UNGeZUXeOGQD+pMH
	B/b/hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//hKrNikMLMAkWBEVT
	RUMBAf//h5PNynBlMGMWBHJzZXAxW/+Eop2mVB4wHBYEREdTVAQU8jcfM9k62ITZ9/rM
	gclMpWcOdgH/hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//hKrNikML
	MAkWBEVTRUMBAf//h5uV4GllMGMWBHNlcGkxW/+Eop2mVB4wHBYEREdTVAQUGXxNemq5
	5CeiRsSZTO3nu/PDxkD/hKqtilkLMAkWBEVLRVkBAf//hKrBpE8LMAkWBEVQUk8BAf//
	hKrNikMLMAkWBEVTRUMBAf8EggEAc+Orw3sidPOpMn2lgB519c+LSDTCJAJrBPqjE+o3
	DJ7O0ZnRiYlF3ZRpdTO6qUg+dcrUhXlUlCQZ5n1aGn/s/9A+3mgg/iFk6q4ZrYcbv6zB
	9q1EoM8QXNUS6pZKoQNQFEfabUjq4KfBVgOywLrsQkhXpMH0ynYVjrsZ8wTE1FPc4sp7
	POGkJtxN4On+HkXvkLzszj5CYAboA5AeotMnyWrOOQhdlJZdkjSc2V7Rk2xU7oJE1zQM
	72IVNUTWhwCyYWJ4649fCmvcDf58KWYx3goIA0INz1wckO2F77uelVv6Z8yKV19cTYI5
	OV+A+FT3C5lTULy0y1JPO+4Jr3TY3DCCCNYwggP4MIIC4KADAgECAgEQMA0GCSqGSIb3
	DQEBBQUAMGIxCzAJBgNVBAYTAlVTMRMwEQYDVQQKEwpBcHBsZSBJbmMuMSYwJAYDVQQL
[... 87 lines cut ...]

13:40:22 service.c:185 service_receive_with_timeout(): could not read data
13:40:22 property_list_service.c:190 internal_plist_receive_timeout(): PROPERTY_LIST_SERVICE_E_RECEIVE_TIMEOUT PList

13:40:22 restore.c:146 restored_receive(): RESTORE_E_UNKNOWN_ERROR PList

13:40:22 restore.c:152 restored_receive(): RESTORE_E_PLIST_ERROR PList

13:41:22 service.c:169 service_receive_with_timeout(): SERVICE_E_SUCCESS

13:41:22 service.c:185 service_receive_with_timeout(): could not read data
13:41:22 property_list_service.c:190 internal_plist_receive_timeout(): PROPERTY_LIST_SERVICE_E_RECEIVE_TIMEOUT PList

13:41:22 restore.c:146 restored_receive(): RESTORE_E_UNKNOWN_ERROR PList

13:41:22 restore.c:152 restored_receive(): RESTORE_E_PLIST_ERROR PList

13:42:22 service.c:169 service_receive_with_timeout(): SERVICE_E_SUCCESS

13:42:22 service.c:185 service_receive_with_timeout(): could not read data
13:42:22 property_list_service.c:190 internal_plist_receive_timeout(): PROPERTY_LIST_SERVICE_E_RECEIVE_TIMEOUT PList

13:42:22 restore.c:146 restored_receive(): RESTORE_E_UNKNOWN_ERROR PList

13:42:22 restore.c:152 restored_receive(): RESTORE_E_PLIST_ERROR PList

RonnieLLL, 2015-09-24:
I have resore(iOS 9.0.1) succesed on iPad air2 \ iPhone 5 \ iPhone 5s \ iPhone 6 \ ipad mini2
but iOS 9.0 is failed yet(only iPad), I run it on windows 7. and Mac OX 10.9+。

jalcock0, 2015-09-24:
RonnieLLL have you made any changes to your built of idevicerestore to work with iPad Air?

If so, could you offer a branch of the source, I am failing with iPad Air iOS 9.

FunkyM, 2015-09-24:
Please do not mix different issues within one ticket and rather create a new one for your issue.
This has been fixed in git master. Restore of all current devices and iOS firmwares is working.

closed 2015-09-24 by FunkyM