libimobiledevice/idevicerestore #52: Cant put device into recovery mode from DFU mode on win

xiaojiangDraug, 2015-09-23:
I get this trouble on win7 system. 
logs like this:
"ERROR: Unable to connect to recovery device
ERROR: Unable to place device into recovery mode from DFU mode"
I Check the DeviceIoControl and GetOverlappedResult getlasterror return is 997 , 1167.  tell the device is not connected. I setup a new libusb-win driver,but it not effective. Getting the same error number. Does anyone fall in this problem?

FunkyM, 2015-09-24:
How did you compile? What head revision or version are you using?
Which libirecovery version are you using? Which device are you using?
Please also mind that officially Windows is not supported.

xiaojiangDraug, 2015-09-25:
I was update all form github, compile with mingw on win7.  It also not useful for the iphone4. more discover is that i can restore it in recovery and normal mode, check the win apple driver, its inf  different between libusb-win GUID and usb-tree-view.  i will recheck the win driver and test it

xiaojiangDraug, 2015-09-29:
I trace  the usb packages， all packages are same to the itunes send on win. It's failed down at dfu_client_free. the reset cmd is OK, but the device not quit the dfu mode. 

itunes usb trace:
   9.0  CTL    a1 03 00 00  00 00 06 00                            GET STATUS             379.1.0        
   9.0  IN     00 b8 0b 00  07 00                                  ......                 379.2.0        
   9    RESET                                                                             380.1.0        
   9    NTSTS  c000000e                                            no device              380.2.0        
   9.0  CTL    21 01 00 00  00 00 00 00                            DOWNLOAD FIRMWAR       381.1.0        
   9.0  NTSTS  c000009d                                            dev not present        381.2.0        
  23.0  CTL    80 06 00 01  00 00 12 00                            GET DESCRIPTOR         382.1.0        
  23.0  IN     12 01 00 02  00 00 00 40  ac 05 81 12  00 00 02 03  .......@........       382.2.0        
               04 01                                               ..                     382.2.16       
  23.0  CTL    80 06 04 03  09 04 03 01                            GET DESCRIPTOR         383.1.0        
  23.0  IN     b6 03 43 00  50 00 49 00  44 00 3a 00  38 00 39 00  ..C.P.I.D.:.8.9.       383.2.0        
               33 00 30 00  20 00 43 00  50 00 52 00  56 00 3a 00  3.0. .C.P.R.V.:.       383.2.16       
  23.0  CTL    80 06 00 02  00 00 40 00                            GET DESCRIPTOR         384.1.0        
  23.0  IN     09 02 39 00  02 01 05 80  fa 09 04 00  00 01 fe 01  ..9.............       384.2.0        
               02 00 07 05  04 02 00 02  00 09 04 01  00 00 ff ff  ................       384.2.16       
  23.0  CTL    00 09 01 00  00 00 00 00                            SET CONFIG             385.1.0        
  23.0  CTL    80 06 00 02  00 00 09 00                            GET DESCRIPTOR         386.1.0        
  23.0  IN     09 02 39 00  02 01 05 80  fa                        ..9......              386.2.0        
  23.0  CTL    80 06 00 02  00 00 39 00                            GET DESCRIPTOR         387.1.0        
  23.0  IN     09 02 39 00  02 01 05 80  fa 09 04 00  00 01 fe 01  ..9.............       387.2.0        
               02 00 07 05  04 02 00 02  00 09 04 01  00 00 ff ff  ................       387.2.16       
  23.0  CTL    80 06 01 03  00 00 ff 00                            GET DESCRIPTOR         388.1.0(2)     
  23.0  IN     5e 03 20 00  4e 00 4f 00  4e 00 43 00  3a 00 41 00  ^. .N.O.N.C.:.A.       388.2.0        
               30 00 36 00  45 00 36 00  39 00 35 00  42 00 31 00  0.6.E.6.9.5.B.1.       388.2.16       
  23.0  CTL    00 0b 00 00  00 00 00 00                            SET INTERFACE          390.1.0        
  23.0  USTS   c0000004                                            stall pid              390.2.0        
  23.0  CTL    41 00 00 00  00 00 00 00                            VENDOR                 391.1.0

restore usb trace:
   9.0  CTL    21 01 8d 00  00 00 00 00                            DOWNLOAD FIRMWAR       380.1.0        
   9.0  CTL    a1 03 00 00  00 00 06 00                            GET STATUS             381.1.0        
   9.0  IN     00 32 00 00  06 00                                  .2....                 381.2.0        
   9.0  CTL    a1 03 00 00  00 00 06 00                            GET STATUS             382.1.0        
   9.0  IN     00 b8 0b 00  07 00                                  ......                 382.2.0        
   9    RESET                                                                             383.1.0        
   9    ok                                                                                383.2.0        
   9.0  CTL    21 01 00 00  00 00 00 00                            DOWNLOAD FIRMWAR       384.1.0        
   9    RESET                                                                             385.1.0        
   9    ok                                                                                385.2.0        
   9.0  CTL    80 06 01 03  00 00 ff 00                            GET DESCRIPTOR         386.1.0(2)     
   9.0  IN     5e 03 20 00  4e 00 4f 00  4e 00 43 00  3a 00 32 00  ^. .N.O.N.C.:.2.       386.2.0        
               37 00 45 00  36 00 41 00  34 00 41 00  42 00 44 00  7.E.6.A.4.A.B.D.       386.2.16       
   9.0  CTL    00 0b 00 00  00 00 00 00                            SET INTERFACE          388.1.0  

any suggest to fix it?

xiaojiangDraug, 2015-09-30:
good! I fix it. when sending ibec the data is longer than itunes send on the heard. i fix the data align 64bit. like this 		uint32_t fillsize = 0;
		                        if ((tsize % 0x40) != 0) {
			                     fillsize = ((tsize / 0x40) + 1) * 0x40;
		                        }  
Now , is ok to enter recovery mode.

Tristansibug, 2015-12-24:
would you please upload the whole lbimobiledevice you have? please i am really in need of help

daniel-bluesea, 2016-06-30:
hi xiaojiangDraug, FunkyM

can you please provide full solution?

still encountering the following problem:

...
Sending iBEC (287116 bytes)...
ERROR: Unable to connect to recovery device
ERROR: Unable to place device into recovery mode from DFU mode

nikias, 2016-06-30:
@daniel-bluesea actually the latest code in git should have the necessary fixes. Did you recompile everything? Also are you talking about the same problem (restore from DFU mode)?

daniel-bluesea, 2016-07-04:
@nikias yes i'm talking about restore from dfu mode, and yes i recompiled everything, on windows!, and yes it seems to be working fine :)

thanks!

iCreamFX, 2016-07-29:
@nikias How do you recompile it? I really need to fix my iPhone 4, like really, can you please tell me asap?

nikias, 2016-07-29:
@iCreamFX this is not a support forum. Basic knowledge about compilation etc. is expected to be a prerequisite if you deal with these kind of project. While it kind of depends on the platform the commands to compile it are in the README files. For windows, use mingw.
Anyway, please don't forget that idevicerestore is not a magic tool, it actually doesn't do more than iTunes also does, so if it doesn't work with iTunes, it's extremely unlikely that idevicerestore will change anything about it.