libimobiledevice/idevicerestore #53: No package "openssl" found

kuragi, 2015-09-23:
Hi:

I tried to install idevicerestore on Kali linux, but I run autogen.sh and give me the following error:
--------------------------------------------------------------------------------------------------------------------------------
root@kali:~/Descargas/idevicerestore-master# ./autogen.sh 
libtoolize: putting auxiliary files in `.'.
libtoolize: linking file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: linking file `m4/libtool.m4'
libtoolize: linking file `m4/ltoptions.m4'
libtoolize: linking file `m4/ltsugar.m4'
libtoolize: linking file `m4/ltversion.m4'
libtoolize: linking file `m4/lt~obsolete.m4'
libtoolize: Consider adding `-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
checking whether make supports nested variables... yes
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking target system type... x86_64-unknown-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
/root/Descargas/idevicerestore-master/missing: Unknown `--is-lightweight' option
Try `/root/Descargas/idevicerestore-master/missing --help' for more information
configure: WARNING: 'missing' script is too old or missing
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libirecovery... yes
checking for libimobiledevice... yes
checking for libplist... yes
checking for libzip... yes
checking for libcurl... yes
checking for openssl... no
configure: error: Package requirements (openssl >= 0.9.8) were not met:

No package 'openssl' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables openssl_CFLAGS
and openssl_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
--------------------------------------------------------------------------------------------------------------------------------

I can run openssl on the terminal normally, Kali is installed on a macbook late 2009.

Can helpme please, a can't install idevicerestore

kuragi, 2015-09-23:
This is my version of openssl
--------------------------------------------------------------------------------------------------------------------------------
root@kali:~/Descargas/idevicerestore-master# openssl
OpenSSL> version
OpenSSL 1.0.1k 8 Jan 2015
OpenSSL> 
--------------------------------------------------------------------------------------------------------------------------------

FunkyM, 2015-09-24:
You need to install the development packages for openssl from your distribution.
It is not enough to just install the runtime openssl binaries.
This might help, too:
http://superuser.com/questions/371901/openssl-missing-during-configure-how-to-fix

closed 2015-09-24 by FunkyM

Luiz-Monad, 2020-06-01:
It would be nice if we could also choose to use gnuTLS instead, just like we can with libimobiledevice.

nikias, 2020-06-01:
It's only used for SHA1 calculation, so it can easily be replaced.

Luiz-Monad, 2020-06-01:
I'll have a try then !

nikias, 2020-06-01:
Well I am actually thinking about just adding a sha1.c/h to the project itself.

nikias, 2020-06-02:
@Luiz-Monad check out commit 571e9645410482e74714e5089048682e05b3d3f8. I made OpenSSL optional. Can be disabled with `--without-openssl` and is otherwise auto-detected (no OpenSSL found -> uses internal code).