libimobiledevice/idevicerestore #54: Idevicerestore stuck in waiting for device step

DarkUD, 2015-09-24:
I m trying to restore a firmware for an iphone 4s on ubuntu 14.04. I m using the command as a sudo too but I m always stuck in the waiting for device step and can't go further.  By the way I tried the ideviceinfo but it returns 
No device found, is it plugged in?
Don't know what to do next.

xiaojiangDraug, 2015-09-29:
you can use `lsusb` to make sure the device is attached. the callback func not check the plug event notify. you can using the ubuntu 12.04 to get a try.

DarkUD, 2015-09-29:
Okay just tried again and checked with lsusb and the device was found 
 Apple, Inc. Apple Mobile Device [Recovery Mode]

I will try with ubuntu 12.04 and tell you after.

Jabro, 2015-11-30:
any update on this? I have the same problem.

Also when using iTunes after the waiting for device face I get an error message.

ablakely, 2016-01-16:
I'm having this issue with an iPhone 5 install.
Debug just shows it's attempts to connect, never happens.
Device just times out(?) and reenters the normal recovery mode with connect to iTunes screen.

    Identified device as iPhone5,1
    Latest firmware is iPhone5,1_9.2_13C75_Restore.ipsw
    [...]
    About to restore device... 
    Waiting for device...
    Attempt 1 to connect to restore mode device...
    Attempt 2 to connect to restore mode device...
    Attempt 3 to connect to restore mode device...
    Attempt 4 to connect to restore mode device...
    Attempt 5 to connect to restore mode device...
    [...]
    Attempt 180 to connect to restore mode device...
    ERROR: Unable to connect to device in restore mode
    ERROR: Unable to open device in restore mode
    ERROR: Unable to restore device

### lsusb
        Bus 002 Device 040: ID 05ac:12a8 Apple, Inc.

Sum1n, 2016-08-09:
you can use this video to fix this problem restore idevice on kali linux using libmobiledevice
https://www.youtube.com/watch?v=qRin2ggBGjw

betmenwasdie, 2018-04-01:
@ablakely hi, i just fixed this issues.. just open new terminal, and type ```sudo usbmuxd```

V3ntus, 2021-08-28:
> Fixed by installing latest git versions of the following packages: libirecovery, libimobiledevice libusbmuxd usbmuxd libplist - and then starting usbmuxd before running idevicerestore.
Confirmed, fixed on Debian/ParrotOS using an iPhone 4

zDEFz, 2024-03-16:
> > Fixed by installing latest git versions of the following packages: libirecovery, libimobiledevice libusbmuxd usbmuxd libplist - and then starting usbmuxd before running idevicerestore.
> 
> Confirmed, fixed on Debian/ParrotOS using an iPhone 4

Thanks! That helped me in 2024 :D