libimobiledevice/idevicerestore #58: iPhone 5S fails to restore with iOS 9.0.1

mpenttil, 2015-09-30:
versions (fresh clone from 25/09/2015):
./libirecovery/.git/refs/heads/master 84a03bf674ef96957631d90085f82bf58a3cbd40
./idevicerestore/.git/refs/heads/master 1d2303e42500a66970a056b991b5d191dec9958f
./libimobiledevice/.git/refs/heads/master f268393d4e447ac901879bee751d7350c495fed2
./libplist/.git/refs/heads/master 9ca25d293fe7f8aca8d952fc7bb91464fe2d34ab
./usbmuxd/.git/refs/heads/master 423fb8c0e9750190d2b7f9c306df9efaa7080dbd
./libusbmuxd/.git/refs/heads/master 4d365eefe8255e8f693bce008dc71bf415279c7e

Running on Ubuntu 14.04 x64 with default configurations for the projects.

This is potentially a problem in firmware package BuildManifest.xml: It tells that FDRSupport is not needed and progress stops when device asks it anyway.

Needles to mention that restore works with iTunes.

This same problem is presumably present also with iPad Air 1st gen and iPad mini 2.

jalcock0, 2015-10-05:
This issue has resolved with the solution is forcing FDR support for all devices.  This is tested and working with iPad Air 1, iPhone 5, 6 and 6 Plus.

You can change restore.c to make sure FDR is always activated. It starts in this line:

https://github.com/libimobiledevice/idevicerestore/blob/master/src/restore.c#L1740

just make sure that everything inside the if (b) { } is always executed.

jalcock0, 2015-10-05:
We will test on the iPad Mini 3 and iPhone 5S asap.  Fix courtesy of Nikias

Sum1n, 2016-08-09:
you can use this video to fix this problem restore idevice on kali linux using libmobiledevice
https://www.youtube.com/watch?v=qRin2ggBGjw