libimobiledevice/idevicerestore #61: Unable to connect to device in restore mode - windows

ultra360, 2015-10-11:
Not sure if I can ask for help here but I'll give it a try
Trying to restore ios 8.4 on iphone6 - windows.
Is this possible?

This is what I get

C:\libi>idevicerestore.exe -u xxxxxxxxxxxxxxxxxxxxxx
7,2_8.4_12H143_Restore.ipsw
NOTE: using cached version data
Found device in Recovery mode
Identified device as iPhone7,2
Extracting BuildManifest from IPSW
Product Version: 8.4
Product Build: 12H143 Major: 12
INFO: device serial number is xxxxxxxxxxxx
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID xxxxxxxxxxxxxxxx
Getting ApNonce in recovery mode... fa 86 2e a5 ae 0
 bf 2f e3 45 ad
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 6d 16 0b d6 c3
d 07 1f 3f e4 23
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controll
Sending TSS request attempt 1... Failed to connect t
nection refused
Request URL set to https://17.171.36.30/TSS/controll
Sending TSS request attempt 2... SSL: certificate su
es not match target host name '17.171.36.30'
Request URL set to https://17.151.36.30/TSS/controll
Sending TSS request attempt 3... SSL: certificate su
es not match target host name '17.151.36.30'
Request URL set to http://gs.apple.com/TSS/controlle
Sending TSS request attempt 4... response successful
Received SHSH blobs
Using cached filesystem from 'iPhone7,2_8.4_12H143_R
Extracting iBEC.n61.RELEASE.im4p...
Not personalizing component iBEC...
Sending iBEC (643230 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-2817.1.94
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~iphone.t7000.im4p...
Not personalizing component AppleLogo...
Sending AppleLogo (6657 bytes)...
Extracting 058-15192-139.dmg...
Not personalizing component RestoreRamDisk...
Sending RestoreRamDisk (17481873 bytes)...
Extracting DeviceTree.n61ap.im4p...
Not personalizing component RestoreDeviceTree...
Sending RestoreDeviceTree (162700 bytes)...
Extracting kernelcache.release.n61...
Not personalizing component RestoreKernelCache...
Sending RestoreKernelCache (9454425 bytes)...
About to restore device...
Waiting for device...
ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
ERROR: Unable to restore device

cheers,
ultra

ninosamurai, 2015-10-23:
hello i have the same issue whit iphone 4s, custom firmware, it's now stuck in recovery mode, i tried to compile idevicerestore-master on VS 2013 without luck, for tried to debug the code but some dependencis of the proyect are missing (a lot of link errors), someone have the solution, please help, i have the device for paperweight :)

icehend, 2015-10-27:
i have the same problem
iphone5 7.1.1

mohsin789, 2016-01-18:
same issue for 5s how to solve this issue please help

code-boogie-yeah-groovy, 2016-03-10:
i have the same problem...

code-boogie-yeah-groovy, 2016-03-10:
i have the same problem with iphone 5.2 ios9,2.1(custom)

iCreamFX, 2016-07-28:
We all have the same problem :P If someone tries to help this is what the terminal says:

> C:\libi>idevicerestore.exe -u --DX3HLKB1DP0N -c iPhone3,2_7.1.2_11D257_Restore.i
psw
NOTE: using cached version data
Found device in Normal mode
Identified device as iPhone3,1
Extracting Restore.plist from IPSW
Product Version: 7.1.2
Product Build: 11D257 Major: 11
Device supports Image4: false
Custom firmware requested. Disabled TSS request.
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Extracting filesystem from IPSW
Entering recovery mode...
INFO: device serial number is DX3HLKB1DP0N
Extracting iBEC.n90ap.RELEASE.dfu...
Not personalizing component iBEC...
Sending iBEC (283020 bytes)...
Getting ApNonce in recovery mode... 54 a8 54 31 58 f8 1e ce 88 52 e3 a5 83 49 32
 5b 51 45 3c 05
Recovery Mode Environment:
iBoot build-version=iBoot-1940.10.58
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~iphone.s5l8930x.img3...
Not personalizing component AppleLogo...
Sending AppleLogo (6796 bytes)...
Extracting 058-4266-010.dmg...
Not personalizing component RestoreRamDisk...
Sending RestoreRamDisk (9584908 bytes)...
Extracting DeviceTree.n90ap.img3...
Not personalizing component RestoreDeviceTree...
Sending RestoreDeviceTree (74764 bytes)...
Extracting kernelcache.release.n90...
Not personalizing component RestoreKernelCache...
Sending RestoreKernelCache (8115660 bytes)...
About to restore device...
Waiting for device...
ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
ERROR: Unable to restore device