libimobiledevice/idevicerestore #63: Unable to restore device

cusmire, 2015-10-12:
Hello guys! Im unable to restore my idevice.
NOTE: using cached version data
Found device in Recovery mode
Identified device as n42ap, iPhone5,2
Extracting Restore.plist from IPSW
Product Version: 9.0
Product Build: 13A344 Major: 13
INFO: device serial number is DQGMN1MYF39D
Device supports Image4: false
Custom firmware requested. Disabled TSS request.
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Using cached filesystem from '/media/philicien/01D060E966B29350/lib/custom/058-07912-297.dmg'
Extracting iBEC.n42ap.RELEASE.dfu...
ERROR: zip_name_locate: Firmware/dfu/iBEC.n42ap.RELEASE.dfu
ERROR: Unable to extract iBEC.n42ap.RELEASE.dfu from /media/philicien/01D060E966B29350/lib/custom.ipsw
ERROR: Unable to extract component: iBEC
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC
philicien@philicien-Latitude-E4300:~$ sudo idevicerestore -c -u 32ec4620e59ca3c66e9fe2f12f0ac5a7a204073d  '/media/philicien/01D060E966B29350/lib/custom.ipsw'
NOTE: using cached version data
Found device in Normal mode
Identified device as n42ap, iPhone5,2
Extracting Restore.plist from IPSW
Product Version: 9.0
Product Build: 13A344 Major: 13
Device supports Image4: false
Custom firmware requested. Disabled TSS request.
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Extracting filesystem from IPSW
[==================================================] 100.0%
Entering recovery mode...
INFO: device serial number is F2LL2NHLF8H4
Extracting iBEC.n42ap.RELEASE.dfu...
ERROR: zip_name_locate: Firmware/dfu/iBEC.n42ap.RELEASE.dfu
ERROR: Unable to extract iBEC.n42ap.RELEASE.dfu from /media/philicien/01D060E966B29350/lib/custom.ipsw
ERROR: Unable to extract component: iBEC
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC
Thank you for your comming help

Harizz95, 2016-04-04:
same problem with me ...  :(

nikias, 2017-06-15:
Custom firmware doesn't work for rhe iPhone 5.

closed 2017-06-15 by nikias