libimobiledevice/idevicerestore #65: ERROR: Unable to restore device

Poulpoutine, 2015-10-20:
I have a problem:

[...]

Sending filesystem now...
[==========                                        ]  19.0%ERROR: Unable to send data to ASR. Sent 0 of 20 bytes.
ERROR: Unable to send chunk checksum
ERROR: Unable to send payload to ASR
ERROR: Unable to send filesystem
ERROR: Unable to successfully restore device
Got status message
Status: Fail
Log is available:
 = 0x0 
Scaler 0 enableDeviceClock(false,FULL_CLOCK_INDEX) = 0x0 
void AppleRGBOUT::verify_swap_gated(IOMFBSwapIORequest *, bool *) failed fIsDPOut: 0 fDPDriver: 0 fTVOUTDelegate: 0
AppleRGBOUT verify_swap failed
void AppleRGBOUT::verify_swap_gated(IOMFBSwapIORequest *, bool *) failed fIsDPOut: 0 fDPDriver: 0 fTVOUTDelegate: 0
AppleRGBOUT verify_swap failed
AppleS5L8920XBasebandSPIController::loadConfiguration: NCLK Frequency 48857142, Prescaler 3
BasebandSPIDevice::exitLowPower: Invalid state inactive
AppleSerialMultiplexer: adap::start: Frame size is 2048, Rx Buffer count is 16
AppleSerialMultiplexer: mux::start: created new mux (17) for spi-baseband with adapter BasebandSPIDevice
AppleSerialMultiplexer: debugparams::init: Parsed flags "" ( 0 )
AppleSerialMultiplexer: !! mux::setPowerStateGated: Skipping power state change
AppleSamsungSerial: Identified Serial Port on ARM Device=uart1 at 0x82600000(0x8bf4e000)
AppleSamsungSerial: Identified Serial Port on ARM Device=uart2 at 0x82700000(0x8bf56000)
AppleSamsungSerial: Identified Serial Port on ARM Device=uart3 at 0x82800000(0x8ca26000)
AppleSamsungSerial: Identified Serial Port on ARM Device=uart6 at 0x82b00000(0x8ca28000)
[HPark] AUD10::start: 0x800f6000, highland-park mIISNub: 0x91932080, sampleRate = 44100, ol=10, oi=12
AppleCS42L61Audio: could not allocate control 395
AppleARMBacklight::start: _minDACLevel=456 _maxDACLevel=1971
AppleMultitouchN1SPI: successfully started
AppleRGBOUT: TVOUT device is detected
ApplePinotLCD: _lcdPanelID: 0xa5970dc8 _lcdPanelClass: 2
IOReturn AppleMobileCLCD::set_ditherTable_state(bool), mIsCLCDDitherFor8Bits is true, no dynamic dither table.
[HPark] AUD10::start: 0x800f6000, highland-park mIICNub: 0x8fb5d540, mIISNub: 0x91932080, mSerialNub: 0x918fb400, sampleRate = 44100, ol=10, oi=12
AppleD1815PMUPowerSource: AppleUSBCableDetect 1
AppleD1815PMUPowerSource: AppleUSBCableType USBHost
[HPark] AUD10 ATSc values: [ mic1: 0x00000010, mic2: 0x000000f7, mic3: 0x00000000, recv: 0x00000000, spk: 0x00000000 ]
[HPark] AUD10::setPowerState() function called whereas Start() is not finished
AppleSynopsysOTGDevice::gated_handleUSBCableConnect cable connected, but don't have device configuration yet
AppleSynopsysOTGDevice::gated_handleUSBCableConnect cable connected, but don't have device configuration yet
AppleMultitouchN1SPI: detected HBPP. driver will be kept alive
IOSDIOController::enumerateSlot(): Searching for SDIO device in slot: 0
IOSDIOController::enumerateSlot(): Found SDIO I/O device. Function count(1), memory(0)
IOSDIOIoCardDevice::parseFn0CISTuple(): Device manufacturer ID 0x2d0, Product ID 0x4329
IOSDIOIoCardDevice::parseFn0CISTuple(): Manufacturer: ""
IOSDIOIoCardDevice::parseFn0CISTuple(): Product:      ""
IOSDIOIoCardDevice::parseFn0CISTuple(): ProductInfo0: "s=B1"
IOSDIOIoCardDevice::parseFn0CISTuple(): ProductInfo1: "P=N90 m=3.1 V=m"
AppleBCMWLANBusInterfaceSdio::start(): Started by: IOSDIOIoCardDevice, AppleBCMWLANV2-259 Jun 11 2014 20:13:01
AppleBCMWLANCore::init(): IO80211-46 Jun 11 2014 20:10:01
000060.775046 wlan.N[0] AppleBCMWLANCore::start(): Starting with MAC Address: 5c:59:48:0f:50:44
IO80211Peer::addPhyStatistics Phy stats started for addr 00:be:ef:ff:ff:ff
IO80211Peer::addPhyStatistics Tx Completion stats started for addr 00:be:ef:ff:ff:ff
IO80211PeerManager::initWithInterface cant add monitoring timer
Started logging for intf
IO80211PeerManager::initWithInterface: inited peer manager
IO80211Interface::init peerManager=0x919db000
start:156
start:156
000060.778701 wlan.N[1] AppleBCMWLANCore::setPowerStateGated():   powerState 1, fStateFlags 0x20, dev 0x802fd000 (this 1, provider 0)
000060.778744 wlan.N[2] AppleBCMWLANCore::setPowerStateGated():  Received power state change before driver has initialized, ignoring
AppleSynopsysOTGDevice::gated_handleUSBCableConnect cable connected, but don't have device configuration yet
read new style signature 0x43313131 (line:389)
[FTL:MSG] VSVFL Register  [OK]
[WMR:MSG] Metadata whitening is set in NAND signature
[FTL:MSG] VFL Init            [OK]
display-scale = 2
[... 273 lines cut ...]
ASR RESTORE PROGRESS: 18%
void AppleUSBDeviceMux::handleMuxTCPInput(mbuf_t) received reset, closing 0x931180b0
ASR: asr: 
ASR: Socket unexpectedly closed while trying to read() offset 0xDBB466C
ASR: __decompressChunk: backing store read at offset 228366682 length 230024 failed with error 5
ASR: asr: 
ASR: __decompressChunk() error: 5
ASR: Failed to read the stream
ASR: : 
ASR: Input/output error
ASR: __decompressChunk: backing store read at offset 227641219 length 235574 failed with error 5
ASR: __decompressChunk: backing store read at offset 226029867 length 244217 failed with error 5
ASR: __decompressChunk: backing store read at offset 226727922 length 210638 failed with error 5
ASR: __decompressChunk() error: 5
ASR: __decompressChunk() error: 5
ASR: __decompressChunk() error: 5
ASR STATUS: fail
ASR: newfs_hfs: MarkExtentUsed::pread(9, 0x27d5a610, 8192, 4096): Invalid argument
ASR: /sbin/newfs_hfs failed with error 256
ASR: Could not restore - Input/output error
ASR STATUS: fail

ERROR: Unable to restore device

Please help me

GeoSn0w, 2016-01-06:
Your IPSW might be broken or even the data cable.

nikias, 2017-06-15:
Could also be a problem with large file support. Closing since no info about device, platform, and command line arguments were given.

closed 2017-06-15 by nikias

GeoSn0w, 2017-06-18:
I've seen the error before on previous versions of idevicerestore, especially the one compiled by quamotion somewhere in ~2015, the issue has been fixed / has stopped appearing upon using an updated version compiled from here. Since the ticket is from 2015, it might be that. The problem used to appear on 5S if I remember well (might have been present on other devices as well).

Conclusion: the pre-compiled binaries are most likely old / outdated and cause problems.

JuaneteMii, 2024-12-07:
I don’t know if this is relevant, but if anyone is here with the same problem, check the storage on your pc. You should have enough free space for the task you’re performing, otherwise everything will go on ignoring the fact you don’t have the file you need.