libimobiledevice/idevicerestore #66: Couldn't restore iPad 2 to iOS 9 - FDR fails.

edgardz, 2015-10-20:
Upgrading an iPad 2 (4.3.5) to iOS 9. Tried with iTunes with no success, always returns error 9. 
A friend told me about idevicerestore and I had no success either. Here is the log:

```
$ sudo idevicerestore -e ~/Library/iTunes/iPad\ Software\ Updates/iPad2\,1_9.0.2_13A452_Restore.ipsw
NOTE: using cached version data
Found device in Recovery mode
Identified device as k93ap, iPad2,1
Extracting BuildManifest from IPSW
Product Version: 9.0.2
Product Build: 13A452 Major: 13
INFO: device serial number is XXXXXXXXXX
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID XXXXXXXXXXXX
Getting ApNonce in recovery mode...
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode...
WARNING: Unable to find BbChipID node
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Using cached filesystem from '/Users/edgardz/Library/iTunes/iPad Software Updates/iPad2,1_9.0.2_13A452_Restore/058-07911-225.dmg'
Sending APTicket (2710 bytes)
Extracting iBEC.k93.RELEASE.dfu...
Personalizing IMG3 component iBEC...
reconstructed size: 297498
Sending iBEC (297498 bytes)...
Getting ApNonce in recovery mode...
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode...
WARNING: Unable to find BbChipID node
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Sending APTicket (2710 bytes)
Recovery Mode Environment:
iBoot build-version=iBoot-2817.1.94
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo.s5l8940x.img3...
Not personalizing component AppleLogo...
Sending AppleLogo (5388 bytes)...
ramdisk-size=0x4000000
Extracting 058-03653-290.dmg...
Not personalizing component RestoreRamDisk...
Sending RestoreRamDisk (20242700 bytes)...
Extracting DeviceTree.k93ap.img3...
Not personalizing component RestoreDeviceTree...
Sending RestoreDeviceTree (78348 bytes)...
Extracting kernelcache.release.k93...
Not personalizing component RestoreKernelCache...
Sending RestoreKernelCache (10445836 bytes)...
About to restore device...
Waiting for device...
Device ffffffffffffffffffffffffffffffff00000019 is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 13
Device ffffffffffffffffffffffffffffffff00000019 has successfully entered restore mode
Hardware Information:
BoardID: 4
ChipID: 35136
UniqueChipID: XXXXXXXXXXXX
ProductionMode: true
Panic log is available:
panic(cpu 0 caller 0x98ac6fa3): "EffacingMediaFilter: initCrypto() failed\n"
Debugger message: panic
OS version: 13A452
Kernel version: Darwin Kernel Version 15.0.0: Thu Aug 20 13:11:09 PDT 2015; root:xnu-3248.1.3~1/RELEASE_ARM_S5L8940X
Paniclog version: 3
Kernel slide:    0x0000000018000000
Kernel text base: 0x98001000
  Boot    : 0x56266ecd 0x00000000
  Sleep  : 0x00000000 0x00000000
  Wake    : 0x00000000 0x00000000
  Calendar: 0x56266eed 0x000d55dc
Panicked task 0x80045778: 734 pages, 1 threads: pid 11: fsck_hfs
panicked thread: 0x80147000, backtrace: 0x8eeb3890
  0x980c9bff
  0x980c9ed5
  0x98020835
  0x98ac6fa3
  0x986a3037
  0x9810e33d
  0x980ddedf
  0x98256c3b
  0x9825a5f3
  0x980eb21b
  0x982b55c9
  0x982b4d55
  0x9830301b
  0x980c6360
Task 0x80046ac8: 6769 pages, 106 threads: pid 0: kernel_task
Task 0x80046790: 604 pages, 3 threads: pid 1: launchd
Task 0x80046458: 602 pages, 6 threads: pid 4: syslogd
Task 0x80046120: 1513 pages, 1 threads: pid 5: restored_externa
Task 0x80045de8: 4352 pages, 4 threads: pid 6: restored_externa
Task 0x80045778: 734 pages, 1 threads: pid 11: fsck_hfs
Starting FDR listener thread
ERROR: Unable to connect to FDR client (-2)
ERROR: Failed to start FDR Ctrl channel
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Partition NAND device (28)
 ```
When it reaches this step, the devices reboots and the script freezes.

PorcupineTheory, 2015-11-22:
I am experiencing the exact same thing. Additionally, at the time of the reboot/script freeze, I get a popup saying "Unable to mount iPhone/Unhandled Lockdown error (-256)"

boboberlin, 2015-11-27:
I´ve the same problem. I want to update ios 4.3.5 to ios 9.2, give me the same error Partition NAND device (28) and Itunes error (9) with my ipad 2 wifi.
Any solutions yet?

PorcupineTheory, 2015-11-27:
Nothing yet.

pekn, 2016-01-05:
You cannot upgrade ancient iPad 2 using iTunes (or libimobiledevice) to iOS 9 (well, assuming Apple doesn't release a fix). You should be able to find discussion threads in Apple support forums easily. You can try to update first to newer version using device itself.

PorcupineTheory, 2016-01-05:
Unfortunately for me (and probably the OP) my device is disabled, or that's
the first thing I would have tried.

pekn, 2016-01-05:
You can try running 'irecovery -n' (in Linux, probably requires sudo). irecovery is a tool from libirecovery and -n option causes the tool to try to leave recovery mode, i.e. boot to normal mode. My experience with other iOS devices - reddish itunes logo: irecovery -n won't work, normal blueish itunes logo: irecovery -n should work.

PorcupineTheory, 2016-01-05:
That'll exit recovery, but it doesn't unlock a disabled device.

boboberlin, 2016-01-13:
So, I´'ve found a Solution to update to IOS 9.So just keep updating to the new Ios 9.3 BETA1 Firmware and you are good to go from iOS 4.3.5.Good Luck.....
It works!!!
Regards,
Burak

PorcupineTheory, 2016-01-13:
That'd be great, but it appears tricky finding the ipsw right now. The only
public facing site is some French site that downloads at 50KB/s. Since my
device is disabled I can't enroll it in the official beta.

PorcupineTheory, 2016-01-14:
That worked. I had to find a direct download link for the ipsw.
https://ghostbin.com/paste/spdu5

nikias, 2017-06-15:
Seems like a defective unit to me since iTunes also fails to restore. Closing.

closed 2017-06-15 by nikias

PorcupineTheory, 2017-06-17:
You didn't even read the whole thread. The problem is solved. The unit was
fine.

nikias, 2017-06-17:
I did read the whole thread and the output
```
panic(cpu 0 caller 0x98ac6fa3): "EffacingMediaFilter: initCrypto() failed\n"
```
and panicked task ```fsck_hfs``` made me assume something is wrong with the device hardware. Repetitive restoring of whatever firmware didn't really sound like a fix to me either.

PorcupineTheory, 2017-06-17:
The fix was the more recent firmware, as the upgrade progression was
locked.

nikias, 2017-06-17:
Thanks for the clarification. "So just keep updating to the new Ios 9.3 BETA1 Firmware" sounded like it only works when trying multiple times. Wasn't clear from @boboberlin's statement.