libimobiledevice/idevicerestore #67: ERROR: could not retrieve device serial number

drnfl, 2015-10-21:
Issue with downgrade to 6.1.3 iphone 4,1 came with ios 8 (build december 2014)
with this message : ERROR: could not retrieve device serial number

denemetahtasi, 2015-10-29:
i have same problem. do you have any idea about this

esamorokov, 2018-04-25:
Could you add a flag to add the serial number, so that the function which is dependent on it would take it. Does it make sense?

paco3346, 2018-11-14:
I'm having this issue with an iPad given to me that's stuck in DFU mode. iTunes can't restore it (and of course doesn't give any helpful information as to why either). I'd like to see what this does if I can skip the serial number check.

esamorokov, 2018-11-14:
It does nothing, I tried it.

paco3346, 2018-11-14:
Bummer, thanks for confirming.

Anyone else that ends up here from Google: I ended up cloning the repo and tweaking things to see how far I could get. After skipping the serial check it'll next get stuck at an error stating it can't put the device into recovery mode.

cmdpwnd, 2019-07-25:
bump.. @paco3346 @nikias 

Anyone find a repeatable workaround? I'm also erring out here.

paco3346, 2019-07-25:
Sorry, no luck here. I ended up concluding that my device was just dead.

wow-hello, 2023-10-22:
Hello! I am trying to downgrade my iPhone3,1 from iOS 7.1.2 to 5.1.1 (jailbroken), but I keep getting the error: "could not retrieve device serial number. Can't continue." iTunes also shows up with "Serial Number: n/a" when in DFU mode. Does anyone know how I could fix this or even possibly get around it? Please let me know!

Device: 2010 MacBook
OS: macOS High Sierra 10.13.6
Processor: 2.4 GHz Intel Core 2 Duo
RAM: 2 GB 1067 MHz DDR3
Graphics: NVIDIA GeForce 320M 256 MB