libimobiledevice/idevicerestore #68: ERROR: Unable to get SHSH blobs for this device

toni90, 2015-10-21:
root@kali:~/idevicerestore# idevicerestore -e /root/Desktop/iPhone3,1_7.1_11D169_Restore.ipsw
NOTE: using cached version data
Found device in Normal mode
Identified device as n90ap, iPhone3,1
Extracting BuildManifest from IPSW
Product Version: 7.1
Product Build: 11D169 Major: 11
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 437693459758
Getting ApNonce in normal mode... ef d5 d9 00 b3 6d ed 05 f9 23 53 21 bd b0 9b c7 89 50 62 c8 
Trying to fetch new SHSH blob
Unable to get SEPNonce from lockdownd
Getting SepNonce in normal mode... failed
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

toni90, 2015-10-21:
root@kali:~/idevicerestore# idevicerestore -d -s -e /root/Desktop/iPhone3,1_7.1_11D169_Restore.ipsw 
Reading data from version.xml
NOTE: using cached version data
Found device in Normal mode
Identified device as n90ap, iPhone3,1
Extracting BuildManifest from IPSW
Product Version: 7.1
Product Build: 11D169 Major: 11
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Getting device's ECID for TSS request
Found ECID 437693459758
Getting ApNonce in normal mode... ef d5 d9 00 b3 6d ed 05 f9 23 53 21 bd b0 9b c7 89 50 62 c8 
Trying to fetch new SHSH blob
Unable to get SEPNonce from lockdownd
Getting SepNonce in normal mode... failed
NOTE: Unable to find BbProvisioningManifestKeyHash node
NOTE: Unable to find BbActivationManifestKeyHash node
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbFDRSecurityKeyHash node
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Applying restore request rules for entry LLB
DEBUG: Applying restore request rules for entry RecoveryMode
DEBUG: Applying restore request rules for entry RestoreDeviceTree
DEBUG: Applying restore request rules for entry RestoreKernelCache
DEBUG: Applying restore request rules for entry RestoreLogo
DEBUG: Applying restore request rules for entry RestoreRamDisk
DEBUG: Applying restore request rules for entry iBEC
DEBUG: Applying restore request rules for entry iBSS
DEBUG: Applying restore request rules for entry iBoot
common.c:printing 4906 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>@Locality</key>
	<string>en_US</string>
	<key>@HostPlatformInfo</key>
	<string>mac</string>
	<key>@VersionInfo</key>
	<string>libauthinstall-293.1.16</string>
	<key>@UUID</key>
	<string>247B0388-39E4-121C-3EFB-2E49FCDB78CF</string>
	<key>ApECID</key>
	<integer>437693459758</integer>
	<key>UniqueBuildID</key>
	<data>
	RztmRK5kC1U4amrtwKDyivn+wBc=
	</data>
	<key>ApChipID</key>
	<integer>35120</integer>
	<key>ApBoardID</key>
	<integer>0</integer>
	<key>ApSecurityDomain</key>
	<integer>1</integer>
	<key>AppleLogo</key>
	<dict>
		<key>Digest</key>
		<data>
		lArhDoO9xzyq+LTg8dJByfxu9X4=
[... 217 lines cut ...]
		<key>ModemStack-Digest</key>
		<data>
		ebeLNZ/4pv2VaNmLFpZEpTQNX0Q=
		</data>
		<key>ModemStack-Length</key>
		<string>0x006f7248</string>
		<key>ModemStack-SecPackDigest</key>
		<data>
		Q9DHqpgfy1IP1vi44STUTiNpnqY=
		</data>
		<key>RamPSI-PartialDigest</key>
		<data>
		QAQAAMD5AABaj5grvd5JETnsAKZKAD/cTGrVqA==
		</data>
		<key>RamPSI-Version</key>
		<string>0x0002000d</string>
	</dict>
</dict>
</plist>
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.

ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.
)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

DiamondGeezer, 2015-11-24:
I'm having this same error on MBP running Yosemite and TinyUmbrella v7.0.2

nikias, 2015-11-24:
You do realize that idevicerestore can only restore iOS versions that are still signed by Apple (except for devices that have a boot level exploit like iPhone 4 and below)?

DiamondGeezer, 2015-11-25:
Even when attempting to restore using the most recent 9.1 firmware, running: 
idevicerestore -u —<MyUDID> -e /Users/me/Downloads/iPhone7\,1_9.1_13B143_Restore.ipsw **from normal mode results in:**

NOTE: using cached version data
ERROR: normal_idevice_new: can't open device with UDID —<MyUDID>
ERROR: Unable to discover device mode. Please make sure a device is attached.

**This is very strange because running *ideviceinfo* prints out all the correct info.
from recovery mode results in:**

NOTE: using cached version data
Found device in Recovery mode
Identified device as n56ap, iPhone7,1
Extracting BuildManifest from IPSW
Product Version: 9.1
Product Build: 13B143 Major: 13
INFO: device serial number is <MySerialNumber>
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID <MyECID>
Getting ApNonce in recovery mode... 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 
WARNING: Unable to find BbSkeyId node
ERROR: Unable to find required ApNonce in parameters
ERROR: Unable to add img4 tags to TSS request
ERROR: Unable to get SHSH blobs for this device

**and from DFU mode results in:**
 
NOTE: using cached version data
Found device in DFU mode
Identified device as n56ap, iPhone7,1
Extracting BuildManifest from IPSW
Product Version: 9.1
Product Build: 13B143 Major: 13
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID <MyECID>
Getting ApNonce in dfu mode... 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... 
WARNING: Unable to find BbSkeyId node
ERROR: Unable to find required ApNonce in parameters
ERROR: Unable to add img4 tags to TSS request
ERROR: Unable to get SHSH blobs for this device

Compiled on Yosemite yesterday via: brew install libimobiledevice

18chablis, 2016-09-18:
NOTE: using cached version data
ERROR: Unable to discover device mode. Please make sure a device is attached
please any help ..... please how can i fix it!!!!!!!!!!!!!!!

nikias, 2017-06-15:
I think the code wasn't up to date. Closing due to inactivity.

closed 2017-06-15 by nikias