libimobiledevice/idevicerestore #69: LIBUSB_ERROR_ACCESS

matteyeux, 2015-10-28:
After compiled libimobildevice, I have this error Ubuntu 14.04 LTS : <br>
{% highlight c %} 
opening device 05ac:1227...
irecv_open_with_ecid: can't connect to device: LIBUSB_ERROR_ACCESS
{% endhighlight %}

nikias, 2016-07-11:
@matteyeux have you tried running it as root or assigning different permissions to the usb device file? I am actually using a udev rules file for that. Given that my local user is in the group "plugdev",  the following udev rule does the trick:
```
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="05ac", ATTR{idProduct}=="12[28][0-37]", OWNER="root", GROUP="plugdev", MODE="0660"
```

nikias, 2017-06-15:
libirecovery commit https://github.com/libimobiledevice/libirecovery/commit/b1bc747824c21749d3d09c1aa6b1ce71087cd8cd will make it possible for non-root users to access the usb device.

closed 2017-06-15 by nikias

matteyeux, 2017-07-03:
Thanks ! 
(sorry for late reply)