libimobiledevice/idevicerestore #70: Unable to restore device on Windows

matteyeux, 2015-11-10:
[image]

FunkyM, 2015-11-10:
Please provide background information about your build.
* Which platform, architecture?
* How did you build it?
* Did you modify the code, can you reference your sources?
* What versions of the libimobiledevice stack did you use?
* iOS version used?
* Command-line used to run?
Thanks.

elrhk, 2015-11-10:
Hello, Thank you for your response (matteyeux it's my friends for testing )

1) 64bit Windows 10 platform
2) Construction in Mingw32 with Msys2
3) libimobiledevice 's on your github
4) command line " idevicerestore -d" file.ipsw "
desoler for my poor English I remain at your disposal for more information . I had already compile with older version idevicerestore and no such error . Thank you

elrhk, 2015-11-11:
Hello after several test , I realized that the problem just libzip- 4.dll I will test recompiled with an older version.

aymenim, 2015-11-29:
Hello, 
I am having the same issue as well,  can you please explain, why you believe  libzip- 4.dll is the cause, for me i tried it on windows 10, build using mingw32  updating iPhone 6 to iOS 9.1.

idevicerestore -l -u <device udid> -C <cache dir>

Thankyou

matteyeux, 2015-11-29:
Hi here is a setup  wizard for Libimobiledevice built by [@Dev__Jam](https://twitter.com/Dev__Jam)

closed 2015-12-22 by matteyeux

jackczhang, 2016-04-11:
@matteyeux @elrhk  I got the same error, do you get the the right version of libzip?

elrhk, 2016-04-11:
Version fix for Windows here https://mega.nz/#!g0N11B6Q!OD9vIsOsmy6IE0sxpV_B-rxWwsIPOt87AtgVePFWxSY

jackczhang, 2016-04-11:
@elrhk the version you provide does not fix this issue, i still get the error

About to send BasebandData...
WARNING: Unable to find BbSkeyId node
Sending Baseband TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received Baseband SHSH blobs
ERROR: could not close and write modified archive: Renaming temporary file failed: No such file or directory
ERROR: Unable to send baseband data
ERROR: Unable to successfully restore device
ERROR: Unable to restore device 

there are two libzip.dll, libzip-2.dll does not support large file (>2GB), i rename libzip-4.dll to libzip-2.dll and get this error.
Test with the latest IPSW file iPhone7,1_9.3_13E233_Restore.ipsw, any idea?