libimobiledevice/idevicerestore #71: ERROR: zip_open while trying to restore iPhone 6S plus

daniel-bluesea, 2015-11-12:
i built idevicerestore.exe on mingw.
i used it to successfully restore an iPhone 6S.

but when i tried to restore iPhone 6S plus i got the following errors:

C:\MinGW\bin>idevicerestore.exe "iPhone8,2_9.1_13B143_Restore.ipsw"
NOTE: using cached version data
Found device in Normal mode
Identified device as n66map, iPhone8,2
Extracting BuildManifest from IPSW
ERROR: zip_open: q:\danielsh\mac\iPhone8,2_9.1_13B143_Restore.ipsw: 19
ERROR: zip_open: q:\danielsh\mac\iPhone8,2_9.1_13B143_Restore.ipsw: 19
ERROR: Invalid archive
ERROR: Unable to extract BuildManifest from q:\danielsh\mac\iPhone8,2_9.1_13B143_Restore.ipsw. Firmware file might be corrupt.

btw, when building idevicerestore on mac, i manged to restore iPhone 6S plus, so i dont think the problem is the ipsw file.

closed 2015-11-29 by daniel-bluesea

Charyas, 2016-04-19:
I get the same problem, do you solve it？

alexsoft1, 2016-10-17:
Who will answer this question ?

daniel-bluesea, 2016-10-18:
i think it is already fixed.. it was an issue of "big" ipsw files in a mingw environment

nikias, 2016-10-18:
@daniel-bluesea is right, the issue occurs when libzip on mingw isn't built properly with support for large files (> 2GB)

timonline, 2017-02-25:
@nikias i understand this is old but am facing this issue, see below:

ERROR: zip_open: iPhone_4.0_64bit_10.2_14C92_Restore.ipsw: 11
ERROR: zip_open: iPhone_4.0_64bit_10.2_14C92_Restore.ipsw: 11
ERROR: Invalid archive
[Error] ERROR: Unable to extract BuildManifest from iPhone_4.0_64bit_10.2_14C92_Restore.ipsw. Firmware file might be corrupt.
Done: restoring failed.
Failed with errorcode=-3

Any help?

byalex3gs, 2018-04-24:
yo tengo el mismo error al querer restaurar un iphone 5s alguna solucion

NoDaa, 2018-12-20:
Tengo el mismo problema, alguien puede ayudar?