libimobiledevice/idevicerestore #72: Unable to extract component: iBSS

mec1, 2015-11-17:
I'm new to this and don't know if I didn't set something up right or what I'm trying to do is just not possible. I"m trying to go from 9.0.2 (13A452) to 8.4.1 (12H321). After I attempt this, it errors out at extracting iBBS and locks the phone in recovery mode. The only way out of recovery mode is by using FonePaw to get out. I need some advice.   --Thanks to anyone who can help me out--

C:\mylib>idevicerestore.exe -u 3899A73153E05B0B5AD609CB45D3C727812E3037 -c iPhon
e7,2_8.4.1_12H321_Restore.ipsw
NOTE: using cached version data
Found device in Normal mode
Identified device as iPhone7,2
Extracting Restore.plist from IPSW
Product Version: 8.4.1
Product Build: 12H321 Major: 12
Device supports Image4: true
Custom firmware requested. Disabled TSS request.
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Using cached filesystem from 'iPhone7,2_8.4.1_12H321_Restore/058-24566-023.dmg'
Extracting iBSS.n61ap.RELEASE.dfu...
ERROR: zip_name_locate: Firmware/dfu/iBSS.n61ap.RELEASE.dfu
ERROR: Unable to extract iBSS.n61ap.RELEASE.dfu from iPhone7,2_8.4.1_12H321_Rest
ore.ipsw
ERROR: Unable to extract component: iBSS
ERROR: Unable to send iBSS to device
ERROR: Unable to place device into recovery mode from DFU mode

bullionmotik, 2016-02-17:
I have same issue and doesn't have a solution yet... so frustrating

nikias, 2016-02-17:
Your libzip is probably too old (not compatible with files >= 2GB) or improperly built on windows. Apart from that please be aware that you can't downgrade the firmware anyway (unless it's one of the old devices like iPhone 4 that have a bootrom exploit).

bullionmotik, 2016-04-26:
thanks bro

AlfredGao, 2016-06-27:
Hey bro. Do you use the OS X EI Capitan to do this? I faced this problem and switch to a Yoesmite machine and make it.

Sum1n, 2016-08-09:
you can use this video to fix this problem restore idevice on kali linux using libmobiledevice
https://www.youtube.com/watch?v=qRin2ggBGjw

carlosnewmusic, 2018-01-05:
[image]

vadimszzz, 2022-04-23:
```
x@MacBook-Pro-Vadim ~ % idevicerestore -c /Users/x/Desktop/iPhone9,1_10.0.1_14A403_Restore.ipsw
Found device in Normal mode
Identified device as d101ap, iPhone9,3
Extracting Restore.plist from IPSW
Product Version: 10.0.1
Product Build: 14A403 Major: 14
Device supports Image4: true
Custom firmware requested. Disabled TSS request.
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
################################ [ WARNING ] #################################
# You are trying to DOWNGRADE a 14.6 device with an IPSW for 10.0.1 while    #
# trying to preserve the user data (Upgrade restore). This *might* work, but #
# there is a VERY HIGH chance it might FAIL BADLY with COMPLETE DATA LOSS.   #
# Hit CTRL+C now if you want to abort the restore.                           #
# If you want to take the risk (and have a backup of your important data!)   #
# type YES and press ENTER to continue. You have been warned.                #
##############################################################################
> YES
Checking IPSW for required components...
ERROR: iBSS file Firmware/dfu/iBSS.d101ap.RELEASE.dfu not found in IPSW
ERROR: iBEC file Firmware/dfu/iBEC.d101ap.RELEASE.dfu not found in IPSW
ERROR: Could not find all required components in IPSW /Users/x/Desktop/Archive.zip
x@MacBook-Pro-Vadim ~ % >
```