libimobiledevice/idevicerestore #73: iphone 6s plus ipsw file is bigger the max value of off_t (long)

daniel-bluesea, 2015-11-18:
it seems that libzip is expecting off_t max length for ipsw files. off_t is "long" under mingw.
iphone 6s plus has an ipsw file that is larger then max long value (2147483647), hence, idevicerestore fails to open this file.

is there a fix for that?

FunkyM, 2015-11-18:
Yes, your libzip version is probably too old. I've fixed it upstream in libzip and thus not a bug here.
You need to enable large file support and properly enable that when compiling for Windows.

closed 2015-11-18 by FunkyM

daniel-bluesea, 2015-11-24:
thanks for your quick response.

did you compile it for windows?
if i'm using mingw-32 bit i cant enable the large file support..
if i'm using mingw-64 bit i can build libzip (1.0.1) with large file support but other components are having "hard time" to be compiled..

i'm trying to build idevicerestore in order to restore/flash an iphone 6s plus (btw, building idevicerestore for iphone 6s was ok)

nicolassan, 2016-04-05:
I have the same issue for ipsw file larger than 2gb..
I can't Enable large file support with mingw-64 bit (windows)
Do you find a solution daniel ?

daniel-bluesea, 2016-04-10:
yes.

you need to enable (sometimes hard-coded enable) the support for large file in all the idevicerestore dependencies packages