libimobiledevice/idevicerestore #75: DFU mode restore fails on OS X

nikias, 2015-11-24:
On OS X Yosemite and El Capitan (and IIRC even earlier versions) a restore of a device in DFU mode fails in the middle of uploading the iBEC. This ticket is a reminder to myself to look into it but if someone has an idea I'll greatly appreciate it :)

alexxgarci, 2016-05-19:
Will this be fixed some day?

alexxgarci, 2016-05-24:
I tried to restore a device in Mavericks too. It doesn't work!
Any suggestions?

nikias, 2016-05-27:
A small update. I figured out that the reason why it fails is that after uploading the iBSS the device is not reset properly. Right now it is hard to tell what properly means since the code already issues a USB port reset; however the one that the MobileDevice framework does seems to be different. Investigation continues...

nikias, 2016-05-27:
Maybe also the device doesn't accept the iBSS for some reason; when running the Mobile Device framework restore it seems like the device resets on its own while with our code it doesn't... looks like I am missing something.

nikias, 2017-06-15:
This should be fixed by now.

closed 2017-06-15 by nikias

dfear, 2017-10-05:
well.....it isn't, it still happens.

nikias, 2017-10-06:
@dfear it works, I just restored my iPad mini 4 from DFU mode in MacOS Sierra (libirecovery built with IOKit support)