libimobiledevice/idevicerestore #77: Unable to restore iPhone 5s: Waiting for device

Jabro, 2015-11-30:
I'm unable to restore an iPhone 5s using the current git version of idevicerestore:

[...]
Received SHSH blobs
Using cached filesystem from 'iPhone6,2_9.1_13B143_Restore/058-25400-078.dmg'
DEBUG: tss_response_get_path_by_entry: No entry 'iBEC' in TSS response
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.n53.RELEASE.im4p...
Writing data to iBEC
Sending iBEC (644140 bytes)...
opening device 05ac:1281...
found device with ECID 0000050dcac0be10
Setting to configuration 1
Setting to interface 0:0
Recovery Mode Environment:
iBoot build-version=iBoot-2817.10.34
iBoot build-style=RELEASE
Sending AppleLogo...
DEBUG: tss_response_get_path_by_entry: No entry 'AppleLogo' in TSS response
NOTE: No path for component AppleLogo in TSS, will fetch from build_identity
Extracting applelogo@2x~iphone.s5l8960x.im4p...
Writing data to AppleLogo
Sending AppleLogo (11658 bytes)...
ramdisk-size=0x10000000
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreRamDisk' in TSS response
NOTE: No path for component RestoreRamDisk in TSS, will fetch from build_identity
Extracting 058-25363-078.dmg...
Writing data to RestoreRamDisk
Sending RestoreRamDisk (23675936 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreDeviceTree' in TSS response
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.n53ap.im4p...
Writing data to RestoreDeviceTree
Sending RestoreDeviceTree (156090 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.n53...
Writing data to RestoreKernelCache
Sending RestoreKernelCache (11133719 bytes)...
About to restore device... 
Waiting for device...
Attempt 1 to connect to restore mode device...
[...]
Attempt 180 to connect to restore mode device...
ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
ERROR: Unable to restore device

Note: I'm also unable to restore using iTunes. It fails after showing: Wwaiting for IPhone and the Iphone reboots

byexmed, 2015-12-06:
hi haw are you
iPhone6,2_9.1_13B143_Restore -- root filesystem key 
help me pls 
thanks

Sum1n, 2016-08-09:
you can use this video to fix this problem restore idevice on kali linux using libmobiledevice
https://www.youtube.com/watch?v=qRin2ggBGjw