libimobiledevice/idevicerestore #79: idevice does not find the iphone is recovery mode/dfu mode.

mmohamedyaser, 2015-12-31:
idevicerestore/ideviceinfo does not find it. I have updated all the packages. Still no luck. It works in normal mode but once the iphone is sent to recovery mode, gives me an error 
>
ERROR: Unable to connect to device in recovery mode
ERROR: Unable to enter recovery mode
ERROR: Unable to place device into recovery mode from Normal mode 
 
[image]

Healdb, 2016-01-01:
I am also having this problem with my iPad 2 wifi on Windows

mmohamedyaser, 2016-01-04:
Try using this dll file. It might help you out. 
https://mega.nz/#!kURxmSgY!oY7zUJqvaIPPcRFZIWfnTTGHWDUQ9lkM1jP9j_Ef7jo

carlm3, 2016-01-06:
Thanks mmohamedyaser, for trying to help. That file didnt work for me (iPad 2 Wifi).
I'm also having a similar issue, I explain my problem in full here --> https://github.com/libimobiledevice/idevicerestore/issues/80

Please help me guys.

philkingston, 2016-02-16:
You should be calling the command with a `sudo`

Sum1n, 2016-08-09:
you can use this video to fix this problem restore idevice on kali linux using libmobiledevice
https://www.youtube.com/watch?v=qRin2ggBGjw