libimobiledevice/idevicerestore #82: idevicerestore does not work for restore on Cellular iOS 9.2/9.2.1/9.3 Devices

joeltoday, 2016-01-25:
idevicerestore does not work for restore on Cellular iOS 9.2/9.2.1/9.3. It works for wifi devices.. has anyone faced this problem?

Here is what I am getting.

idevicerestore -e -l -u xxxx --cache-path xxx

WARNING: Unable to find BbSkeyId node
ERROR: could not close and write modified archive: Renaming temporary file failed: File exists
NOTE: using cached version data
Found device in Normal mode
Identified device as j97ap, iPad5,2
Latest firmware is iPad5,2_9.2.1_13D15_Restore.ipsw
Verifying iPad5,2_9.2.1_13D15_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 9.2.1
Product Build: 13D15 Major: 13
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 6871471883914278
Getting ApNonce in normal mode... de b0 ed 3b 25 10 2f cc b6 88 3c db b6 f4 ab 83 38 58 69 52 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... ec c1 74 84 15 03 77 6b 5f 98 3b 4a d0 9c 89 32 5e 68 2e 78 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting filesystem from IPSW
Entering recovery mode...
INFO: device serial number is DLXQG19JGHMG
Extracting iBEC.j97.RELEASE.im4p...
Sending iBEC (652809 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-2817.20.26
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~ipad.t7000.im4p...
Sending AppleLogo (23207 bytes)...
ramdisk-size=0x10000000
Extracting 058-32156-015.dmg...
Sending RestoreRamDisk (23918077 bytes)...
Extracting DeviceTree.j97ap.im4p...
Sending RestoreDeviceTree (138870 bytes)...
Extracting kernelcache.release.j97...
Sending RestoreKernelCache (12001700 bytes)...
About to restore device... 
Waiting for device...
Device 4c0cd8260a53903717f0888a61e6f3d1597b6236 is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 13
Device 4c0cd8260a53903717f0888a61e6f3d1597b6236 has successfully entered restore mode
Hardware Information:
BoardID: 10
ChipID: 28672
UniqueChipID: 6871471883914278
ProductionMode: true
Starting FDR listener thread
About to send NORData...
Found firmware path Firmware/all_flash/all_flash.j97ap.production
Getting firmware manifest Firmware/all_flash/all_flash.j97ap.production/manifest
Extracting LLB.j97.RELEASE.im4p...
Extracting iBoot.j97.RELEASE.im4p...
Extracting DeviceTree.j97ap.im4p...
Extracting applelogo@2x~ipad.t7000.im4p...
Extracting recoverymode@2x~ipad-lightning.t7000.im4p...
Extracting batterylow0@2x~ipad.t7000.im4p...
Extracting batterylow1@2x~ipad.t7000.im4p...
Extracting batterycharging0@2x~ipad.t7000.im4p...
Extracting batterycharging1@2x~ipad.t7000.im4p...
Extracting glyphplugin@2x~ipad-lightning.t7000.im4p...
Extracting batteryfull@2x~ipad.t7000.im4p...
Extracting sep-firmware.j97.RELEASE.im4p...
Extracting sep-firmware.j97.RELEASE.im4p...
Sending NORData now...
Done sending NORData
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Partition NAND device (28)
About to send FDR Trust data...
Sending FDR Trust data now...
Done sending FDR Trust Data
Waiting for NAND (29)
Waiting for NAND (29)
Waiting for NAND (29)
Waiting for storage device (11)
Creating partition map (12)
Creating partition map (12)
Creating partition map (12)
Checking filesystems (16)
Checking filesystems (16)
Checking filesystems (16)
Waiting for NAND (29)
Waiting for NAND (29)
Waiting for NAND (29)
About to send filesystem...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
Done sending filesystem
Restoring image (14)
Checking filesystems (16)
Checking filesystems (16)
Checking filesystems (16)
About to send KernelCache...
Extracting kernelcache.release.j97...
Sending KernelCache now...
Done sending KernelCache
Loading kernelcache (27)
Unknown operation (18)
Unknown operation (46)
Unknown operation (46)
Flashing NOR (19)
About to send BasebandData...

ERROR: Unable to send baseband data
ERROR: Unable to successfully restore device
ERROR: Unable to restore device

thebaltee, 2016-02-13:
I'm having a similar issues on 9.2 devices.  I can upgrade older devices and ubuntu can interact with the device, except when it's in recovery mode....

coolhiro, 2016-04-04:
if you are using in windows, convert functions fseek and ftell to _fseeki64 and _ftelli64 in asr.c.

jackczhang, 2016-04-08:
I faced the same issue, does anyone get the solution?

jackczhang, 2016-04-13:
This is the libzip-4.dll issue, you cannot compile using configure, just using cmake.