libimobiledevice/idevicerestore #89: Cannot restore iphone 6s in windows

jackczhang, 2016-03-25:
I got the error as below
ERROR: zip_name_locate: Firmware/dfu/iBEC.n56ap.RELEASE.dfu
ERROR: Unable to extract iBEC.n56ap.RELEASE.dfu from E:\SBGMobileManager\iOS_Firmware\iPhone7,1_9.3_13E233_Restore.ipsw
ERROR: Unable to extract component: iBEC
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC

What's wrong?

coolhiro, 2016-04-04:
i found this situation and solved.
fseek and ftell is problem.
firmware(iOS9.3 of iphone6s) size is bigger than 2^31.
thus, you can use _fseeki64 and _ftelli64.
convert 32bit functions to 64bit functions in asr.c.

jackczhang, 2016-04-05:
@coolhiro Thanks for your replay, but after update the file asr.c function fseek and ftell, the problem still exists, any idea?

coolhiro, 2016-04-05:
sorry, i missed something.

change off_t to uint64_t in asr_send_payload of asr.c.
// default code
//off_t i, length, bytes = 0;
// modified code
uint64_t i, length, bytes = 0;

and change below in asr_handle_oob_data_request of asr.c
// default code
// fseeko(file, oob_offset, SEEK_SET);
// if (fread(oob_data, 1, oob_length, file) != oob_length) {
// modified code
_fseeki64( file, oob_offset, SEEK_SET );
uint64_t read_length;
if ( ( read_length = fread(oob_data, 1, oob_length, file) ) != oob_length ) {

these will help you.

jackczhang, 2016-04-05:
@coolhiro still got the error, when trace the code, found that
the error occurred at restore.c function restore_sign_bbfw() -> zip_close()
// this will write out the modified zip
	if (zip_close(za) == -1) {
		error("ERROR: could not close and write modified archive: %s\n", zip_strerror(za));
		res = -1;
	} else {
		res = 0;
	}
	za = NULL;
	zs = NULL;

coolhiro, 2016-04-05:
try to modify below in ipsw_extract_to_file() of ipsw.c
// default code
//off_t i, bytes = 0;
// modified code
uint64_t i, bytes = 0;

i think zip_close() error has occured because of invalid bbfw file extraction.

nicolassan, 2016-04-05:
I had the same issue, i try to apply the modifications, many thanks  coolhiro !!

jackczhang, 2016-04-06:
@coolhiro thanks, but still get the same error.
It should not be the issue about 64 bit functions for large file.
I think it should be zlib or libzip compile issue, could you share your libzip-4.dll to me?

coolhiro, 2016-04-06:
show me your e-mail please.
i will send libzip .lib file and compilable package.

jackczhang, 2016-04-06:
my email is jack_c_zhang@wistron.com, thanks coolhiro.

nicolassan, 2016-04-06:
Hi,

Can you please send me The dll too ?

My email : nicolassan@outlook.com

Thanks,
Nicolas

jackczhang, 2016-04-12:
Libzip compile issue, use CMake instead configure command.

closed 2016-04-12 by jackczhang

walkman4321, 2016-10-22:
I have also same problem.So i follow every instruction in above post.Still It can't restore firmware to my Iphone 5s. When i again debug code i found that it cannot write data to ".dmg" file .Please help me to figure out this problem

vipinbeni, 2016-10-22:
Hi Sir @FunkyM ,
I also having the same problem.So i follow every instruction in above post.Still It can't restore firmware to my Iphone 5s. When i debug code i found that it cannot validate ASR my function fail in below calling in validation. i am working on ubantu 14.4.2 ,32 bit , i change fopen with fopeno64 but still i have issue on below step.

device_error = idevice_connection_receive(asr->connection, buffer, ASR_BUFFER_SIZE, &size);
if (device_error != IDEVICE_E_SUCCESS) {
error("ERROR: Unable to receive data from ASR\n");
free(buffer);
return -1;
}

while (1) {
if (asr_receive(asr, &packet) < 0) {
error("ERROR: Unable to receive validation packet\n");
return -1;
}

dear sir kindly help me as it work fine on ipsw less then 2 gb but fail on ipsw 2.2 gb for 5S ios 10.2

nikias, 2016-10-22:
@walkman4321 as explained here: https://github.com/libimobiledevice/idevicerestore/issues/18 you need to have a newer version of libzip so it can handle files > 2gb.