libimobiledevice/idevicerestore #91: idevicerestore does not work with certain devices in DFU mode

pekn, 2016-04-01:
It looks like idevicerestore does not work with certain devices (e.g. iPhone 5C). The bug seems to be in dfu_send_component function in dfu.c. There are two problems.

First, the following if statement should be executed.

```
if ((client->build_major > 8) && (client->build_major < 11) && !(client->flags & FLAG_CUSTOM) && (strcmp(component, "iBEC") == 0)) {
```

Should it be something like as follows:

```
if (!client->image4supported && (client->build_major > 8) && !(client->flags & FLAG_CUSTOM) && (strcmp(component, "iBEC") == 0)) {
```

Second, the padding code is incorrect:

```
		if ((tsize % 0x100) != 0) {
			fillsize = ((tsize / 0x100) + 1) * 0x100;
		}
```

At least for devices I tested, ticket must be padded to 0x10 boundary instead of 0x100 (I tested with IPSWs with build_major == 11 and build_major == 13). 

So if you are sure that for (client->build_major > 8) && (client->build_major < 11) IPSWs (I couldn't test these cases), padding must be 0x100, then I suggest using 0x10 when client->build_major >= 11.

With above changes, I was able to use idevicerestore successfully with devices in DFU mode - well, iPhone 5S exception, but DFU mode iPhone 5S cannot be restored even with iTunes.

nikias, 2016-04-12:
I am aware of these bugs and have the same fix implemented but not committed yet, except that I assumed it was a 0x40 padding but in my case (iPod Touch 5th gen iOS 9) it would only pad 4 bytes so 0x10 padding would make perfect sense. I will commit these changes now.

nikias, 2016-04-12:
Actually, I just realized there are two paddings applied. One after the ticket data itself which seems to be indeed 0x40 aligned and another one at the end of the data transfer block. I will check again.

nikias, 2016-04-12:
Ok at the end there is no padding applied, just the CRC padding which is handled by libirecovery already.  And in my case the applied padding (checked with USB dump when iTunes is restoring) is 0x11 bytes, so 0x10 padding doesn't make sense. @pekn What is the ticket size you have?

nikias, 2016-05-26:
I fixed the apticket appending condition and the calculation of the padding size with commit https://github.com/libimobiledevice/idevicerestore/commit/cf1e49a9fd86ae6e1c6855824dac068b417f486c.

closed 2016-05-26 by nikias