libimobiledevice/idevicerestore #94: WARNING: size mismatch when parsing MBN file.

jackczhang, 2016-04-08:
Why i got this warning in latest iOS 9.3, can this cause mbn_update_sig_blob function error or update baseband firmware error?

pekn, 2016-04-11:
I got this kind of error when restoring iPhone 6S. Based on quick look, file format may have changed (well, at least contents of file doesn't seem to match with description in mbn.h).

nikias, 2016-06-30:
Indeed the format seems changed. However I found the same format in iOS 8.4 but never had a baseband update error during restore. Weird.

nikias, 2016-06-30:
I actually found it in even older firmwares. The reason it works is because the size is the file size of the extracted component, regardless of what the incorrectly parsed header says:
https://github.com/libimobiledevice/idevicerestore/blob/master/src/mbn.c#L36

And the signature will just go to the end of the file (overwriting the last size_of_signature bytes):
https://github.com/libimobiledevice/idevicerestore/blob/master/src/mbn.c#L63-L69

nikias, 2016-06-30:
Nevertheless, the format should be updated so we can suppress any error messages or warnings.

nikias, 2016-07-11:
Resolved with commit https://github.com/libimobiledevice/idevicerestore/commit/2f92d5555302a00a5a10154d8413235888d27c7f

closed 2016-07-11 by nikias