libimobiledevice/idevicerestore #95: make and make install is not working for me

debugger22, 2016-04-08:
```
Configuration for idevicerestore 1.0:
-------------------------------------------

  Install prefix: .........: /usr/local

  Now type 'make' to build idevicerestore 1.0,
  and then 'make install' for installation.

debugger22@jarvis:~/workspace/idevicerestore$ [master] sudo make
make  all-recursive
make[1]: Entering directory `/home/debugger22/workspace/idevicerestore'
Making all in src
make[2]: Entering directory `/home/debugger22/workspace/idevicerestore/src'
  CC       idevicerestore-dfu.o
dfu.c: In function ‘dfu_get_cpid’:
dfu.c:243:48: warning: initialization makes pointer from integer without a cast [enabled by default]
  const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
                                                ^
dfu.c:248:21: error: dereferencing pointer to incomplete type
  *cpid = device_info->cpid;
                     ^
dfu.c: In function ‘dfu_get_ecid’:
dfu.c:260:48: warning: initialization makes pointer from integer without a cast [enabled by default]
  const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
                                                ^
dfu.c:265:21: error: dereferencing pointer to incomplete type
  *ecid = device_info->ecid;
                     ^
dfu.c: In function ‘dfu_is_image4_supported’:
dfu.c:278:48: warning: initialization makes pointer from integer without a cast [enabled by default]
  const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
                                                ^
dfu.c:283:21: error: dereferencing pointer to incomplete type
  return (device_info->ibfl & IBOOT_FLAG_IMAGE4_AWARE);
                     ^
dfu.c: In function ‘dfu_get_ap_nonce’:
dfu.c:293:48: warning: initialization makes pointer from integer without a cast [enabled by default]
  const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
                                                ^
dfu.c:298:17: error: dereferencing pointer to incomplete type
  if (device_info->ap_nonce && device_info->ap_nonce_size > 0) {
                 ^
dfu.c:298:42: error: dereferencing pointer to incomplete type
  if (device_info->ap_nonce && device_info->ap_nonce_size > 0) {
                                          ^
dfu.c:299:46: error: dereferencing pointer to incomplete type
   *nonce = (unsigned char*)malloc(device_info->ap_nonce_size);
                                              ^
dfu.c:303:28: error: dereferencing pointer to incomplete type
   *nonce_size = device_info->ap_nonce_size;
                            ^
dfu.c:304:29: error: dereferencing pointer to incomplete type
   memcpy(*nonce, device_info->ap_nonce, *nonce_size);
                             ^
dfu.c: In function ‘dfu_get_sep_nonce’:
dfu.c:317:48: warning: initialization makes pointer from integer without a cast [enabled by default]
  const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
                                                ^
dfu.c:322:17: error: dereferencing pointer to incomplete type
  if (device_info->sep_nonce && device_info->sep_nonce_size > 0) {
                 ^
dfu.c:322:43: error: dereferencing pointer to incomplete type
  if (device_info->sep_nonce && device_info->sep_nonce_size > 0) {
                                           ^
dfu.c:323:46: error: dereferencing pointer to incomplete type
   *nonce = (unsigned char*)malloc(device_info->sep_nonce_size);
                                              ^
dfu.c:327:28: error: dereferencing pointer to incomplete type
   *nonce_size = device_info->sep_nonce_size;
                            ^
dfu.c:328:29: error: dereferencing pointer to incomplete type
   memcpy(*nonce, device_info->sep_nonce, *nonce_size);
                             ^
make[2]: *** [idevicerestore-dfu.o] Error 1
make[2]: Leaving directory `/home/debugger22/workspace/idevicerestore/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/debugger22/workspace/idevicerestore'
make: *** [all] Error 2
```

gvancuts, 2016-12-21:
I'm hitting the same problem. Any idea how we can solve this?

debugger22, 2016-12-21:
I couldn't fix it, maybe the author could help here.

gvancuts, 2016-12-21:
I figured out that I had two version of libirecvoery on my system, one that came from https://github.com/Chronic-Dev/libirecovery and the other one from https://github.com/libimobiledevice/libirecovery. Removing the first one (https://github.com/Chronic-Dev/libirecovery) from my system solved this issue!

cmoura, 2016-12-23:
@gvancuts how did you found and removed the wrong librecovery version?

gvancuts, 2016-12-24:
I had installed it myself under `/usr/local` and found it when I did a search on my entire system for `libirecovery` asI was trying to figure what was going on.