libimobiledevice/idevicerestore #99: idevicerestore.exe always stuck at the same place on 5s - Partition NAND device (28)

loonbg, 2016-04-16:
Hi,

During attempts to upgrade/restore software using the idevicerestore.exe devices on iphone 5s it always stuck on Patition NAND device (28) .

I would add only that iTunes restore iPhone without any problem.

What could be causing this problem? How to solve it?

IOs 9.3

Tnx

bzlai, 2016-05-18:
I have the same issue trying to restore with 9.3.1

Sum1n, 2016-08-09:
you can use this video to fix this problem restore idevice on kali linux using libmobiledevice
https://www.youtube.com/watch?v=qRin2ggBGjw

nikias, 2016-10-04:
@loonbg is this still a problem? Please run idevicerestore with ```-d``` command line switch and post the output.

crendl, 2022-05-05:
i'm facing the same issue, here's my output for an ipad 2 (a1395)
```
Device ffffffffffffffffffffffffffffffff00000001 has successfully entered restore mode
Hardware Information:
BoardID: 4
ChipID: 35136
UniqueChipID: 4155533034447
ProductionMode: true
Starting FDR listener thread
Connecting to FDR client at port 1082
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
ERROR: Unable to connect to FDR client (-2)
ERROR: Failed to start FDR Ctrl channel
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
No data to read (timeout)
No data to read (timeout)
No data to read (timeout)
No data to read (timeout)
No data to read (timeout)
```