libimobiledevice/idevicerestore #101: idevicerestore can't restore iPhone 5c from recovery mode, stops at 50%

gair, 2016-05-13:
I'm attempting to restore from recovery mode with the -e option but it always stops at exactly 50%. This only happens with 5c and they can be restored fine with iTunes.

I tried recovering from DFU mode but idevicerestore is unable to put it back in recovery mode.

Filesystem validated
Sending filesystem now...
[==========================                        ]  50.0%02:57:20 idevice.c:332 internal_connection_send(): ERROR: usbmuxd_send returned -104 (Connection reset by peer)
ERROR: Unable to send data to ASR. Sent 0 of 1450 bytes.
ERROR: Unable to send filesystem payload
ERROR: Unable to send payload to ASR

It then reports AMRError 14 followed by a very long hex dump, ending with:

ASR: __decompressChunk() error: 1000XSTA
ASR: fail
ASR: Could not restore - error 1000
ASR STATUS: fail

nikias, 2016-06-30:
It looks like a communication error to me. Does this happen on the same system you can successfully restore with iTunes (windows / OS X) ?

gair, 2016-07-01:
No, it was happening on Linux. I was never able to discover the root cause but it was resolved after a re-image.

closed 2016-07-01 by gair

JyotiNagpal, 2016-10-23:
@gair  have you got solution of this as i am also face same issue...

gair, 2016-10-24:
As I wrote above, I re-imaged the Linux machine I was using and it resolved the issue so I never identified the exact problem.