libimobiledevice/idevicerestore #102: Require release for homebrew formula

ebekebe, 2016-05-18:
I currently try to make a homebrew formula for this project, but since it has no release yet I am unable to get it accepted. I would appreciate if you could make a 0.0.1 release, so I can continue.

(And great thanks guys for this tool! We intent to use it to update hundreds of iOS devices automatically, but only if we can install it with homebrew :-). )

Blackmamba0, 2016-06-15:
I'm on the same path, I tried to do it by Forking the Git and making a Release but it didn't work, we need a Brew tap for idevicerestore please!!

Austcool-Walker, 2019-06-19:
Yes please add support for this I need this package badly that or macports

FunkyM, 2020-06-15:
Closing as version [1.0.0](/libimobiledevice/idevicerestore/releases/tag/1.0.0) has been released.

closed 2020-06-15 by FunkyM