libimobiledevice/idevicerestore #104: Restoring error asr

colbygamer, 2016-06-02:
recv(12, 4) failed: connection closed
unable to read message size: -1
could not receive message
recv(12, 4) failed: connection closed
unable to read message size: -1
could not receive message
Initialized /dev/rdisk0s1s1 as a 2 GB case-sensitive HFS Plus volume
entering ramrod_probe_media
device partitioning scheme is GPT
find_filesystem_partitions: storage=/dev/disk0s1 system=/dev/disk0s1s1 data=/dev
/disk0s1s2 baseband data= log=
entering mount_partition
executing /sbin/fsck_hfs
journal_replay(/dev/disk0s1s1) returned 0
** /dev/rdisk0s1s1
        Cache size should be multiple of cache block size (currently 32K)
        Using cacheBlockSize=32K cacheTotalBlock=2011 cacheSize=64352K.
   Executing fsck_hfs (version hfs-305.10.1).
** Checking non-journaled HFS Plus Volume.
** Detected a case-sensitive volume.
   The volume name is System
** Checking extents overflow file.
** Checking catalog file.
** Checking multi-linked files.
** Checking catalog hierarchy.
** Checking extended attributes file.
** Checking volume bitmap.
** Checking volume information.
** Trimming unused blocks.
** The volume System appears to be OK.
        CheckHFS returned 0, fsmodified = 0
executing /sbin/mount_hfs
/dev/disk0s1s1 mounted on /mnt1
System mounted read-only
AMSupportPlatformCreateBufferFromNativeFilePath: open failed: No such file or di
rectory
__getFDRSealingMap_block_invoke: Could not load sealing map from '/usr/lib/FDRSe
alingMap.plist'
AMFDRSealingMapGetEntriesForDevice: Could not load sealing map
restored_fdr_initialize returned false
RestoredFDRCreate() returned 0
FDR is not supported on this device
RestoredFDRSaveData: restoredFDR is NULL, not able to preserve FDR data.
RestoredFDRSaveData success
entering check_for_restore_log
restore log does not exist
executing /sbin/fsck_hfs
journal_replay(/dev/disk0s1s2) returned 0
** /dev/rdisk0s1s2
        Cache size should be multiple of cache block size (currently 32K)
        Using cacheBlockSize=32K cacheTotalBlock=2011 cacheSize=64352K.
   Executing fsck_hfs (version hfs-305.10.1).
** Checking Journaled HFS Plus volume.
** Detected a case-sensitive volume.
   The volume name is Data
** Checking extents overflow file.
** Checking catalog file.
** Checking multi-linked files.
** Checking catalog hierarchy.
** Checking extended attributes file.
** Checking volume bitmap.
** Checking volume information.
** Trimming unused blocks.
** The volume Data appears to be OK.
        CheckHFS returned 0, fsmodified = 0
executing /sbin/mount_hfs
mount_hfs: Could not exec re-keying daemon /usr/libexec/rolld: No such file or d
irectory
/dev/disk0s1s2 mounted on /mnt2
entering dump_panic_logs
[... 205 lines cut ...]
ASR RESTORE PROGRESS: 86%
ASR RESTORE PROGRESS: 88%
ASR RESTORE PROGRESS: 90%
ASR RESTORE PROGRESS: 92%
ASR RESTORE PROGRESS: 94%
ASR RESTORE PROGRESS: 96%
ASR RESTORE PROGRESS: 98%
ASR: asr: Image failed signature verification
ASR: asr: Failed to read the stream: Authentication error
ASR: __decompressChunk: backing store read at offset 1362758130 length 88679 fai
led with error 80
ASR: __decompressChunk() error: 80
ASR STATUS: fail
SYSLOG: Jun  2 11:05:54 localhost kernel[0] <Notice>: disk0s1s1: alignment error
.
ASR: newfs_hfs:
ASR: MarkExtentUsed::pread(3, 0x17895000, 8192, 4096): Invalid argument
ASR: /sbin/newfs_hfs failed with error 256
ASR: Could not restore - Authentication error
ASR STATUS: fail

ERROR: Unable to successfully restore device
ERROR: Unable to restore device                                                                                                                                                 

How do i fix this error?

Sn0wDev, 2016-06-28:
erm, if 9.3.1 is still signed, i'm going to assume that your ipsw is corrupted. try redownloading, and if it still doesnt work, put the ipad into dfu or recovery before restoring.
that was what worked for me :D

nikias, 2016-06-28:
This error happened to me when I tried to restore to iOS 10 beta. I committed a fix for that; @colbygamer can you try with the latest code from git?

Sn0wDev, 2016-06-28:
hrmmm. strange. it happened when the ipsw seemed to be corrupt. i redownloaded, put my ipad into recovery, and the restore worked perfectly

colbygamer, 2016-06-29:
@sn0wDew it's a custom firmware restore

nikias, 2016-06-30:
@colbygamer custom firmwares don't work for the iPad 2. It only works for older generation devices iPhone 4 or below, iPad 1.
Closing this tickets since custom firmwares are not supported.

closed 2016-06-30 by nikias