libimobiledevice/idevicerestore #105: Unable to discover device model

Dynat, 2016-06-08:
Using Linux Mint and an iPod Touch 2g MC model i have found myself running into an error that i am not quite sure how to resovle

upon running 
`sudo idevicerestore -c ~/mycustomipsw.ipsw`

I end up with 
`NOTE: using cached version data `
`Found device in DFU mode `
`ERROR: unable to discover device model`

rubenmdh, 2016-06-24:
Same here, I cannot restore my iPod touch 6G (it cannot discover the device model)

Here's the log:
NOTE: using cached version data
Found device in Recovery mode
ERROR: Unable to discover device type

nikias, 2016-06-28:
Can you both make sure that you update libirecovery to latest?

rubenmdh, 2016-06-28:
I have the latest version of libimobiledevice for windows [1.2.0 installer] (quamotion).

Regards!

nikias, 2016-06-28:
You need *libirecovery* latest. Also since you are on windows you might need a recent libusb aswell. Windows is not officially supported, but building all required libraries does work using mingw.