libimobiledevice/idevicerestore #106: Not sending NORData... Quitting...

maenpaa24, 2016-06-14:
Hi! I am new at GitHub. I expose my problem just in case someone can help me.

I am trying to restore ios 9.3.2 in my iPhone 6 through ubuntu 16.04. But when I try to do with the following command:

sudo idevicerestore -e -x /home/user/Desktop/iPhone7,2_9.3.2_13F69_Restore_CFW.ipsw

This is what I get:

NOTE: using cached version data
Found device in Recovery mode
Identified device as n61ap, iPhone7,2
Extracting BuildManifest from IPSW
Product Version: 9.3.2
Product Build: 13F69 Major: 13
INFO: device serial number is 99999999999
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Found ECID 6546546165165415
Getting ApNonce in recovery mode... 41 c5 c6 c3 fe 6b 9f fe 27 21 f1 87 af 4d 12 ee db ea 01 90 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 80 68 bc 40 3d ea d2 c7 78 4e 56 f9 e5 51 83 74 9c f3 fe ec 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting filesystem from IPSW
[==================================================] 100.0%
Extracting iBEC.n61.RELEASE.im4p...
Sending iBEC (652808 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-2817.50.3
iBoot build-style=RELEASE
Sending AppleLogo...
Extracting applelogo@2x~iphone.t7000.im4p...
Sending AppleLogo (12140 bytes)...
ramdisk-size=0x10000000
Extracting 058-37507-072.dmg...
Sending RestoreRamDisk (24204797 bytes)...
Extracting DeviceTree.n61ap.im4p...
Sending RestoreDeviceTree (122870 bytes)...
Extracting kernelcache.release.n61...
Sending RestoreKernelCache (12023173 bytes)...
About to restore device... 
Waiting for device...
Device 54a614ca546b4546c4a56445b45a4ca464acc54b is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 13
Device 54a614ca546b4546c4a56445b45a4ca464acc54b has successfully entered restore mode
Hardware Information:
BoardID: 6
ChipID: 654654
UniqueChipID: 45645654154654654
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
Not sending NORData... Quitting...
Cleaning up...
DONE

Thank you in advance!!!

nikias, 2016-06-14:
Why are you passing -x on the command line? It's supposed to be for older devices where you can install a custom firmware to prevent the baseband upgrade. Also the file name *_CFW.ipsw suggests that you actually try to install a modifed ipsw. However, this doesn't work on anything newer than iPhone 4.

maenpaa24, 2016-06-14:
Yes I am trying to install a modified ipsw. I saw some videos where people say it is possible to overcome the errors that appear when one tries to restore a custom firmware, so I wanted to try.

Thank you for the fast and high quality response.

nikias, 2016-06-30:
Closing since this is not a bug.

closed 2016-06-30 by nikias