libimobiledevice/idevicerestore #108: Waiting for device for ever

Sn0wDev, 2016-06-28:


nikias, 2016-06-28:
What platform are you using? Where is it stalling exactly? You need to provide some details, otherwise nobody can really help you.

Sn0wDev, 2016-06-28:
ok. sorry. im using windows 7 x32, and after saying sending restore kernelcache, it says about to restore device, and then gets stuck there

nikias, 2016-06-28:
Does the device actually enter restore mode? After it shows the Apple logo you will see a spinner at the bottom and then a progress bar should appear. If that doesn't happen, the device doesn't enter restore mode at all so the failure is actually before that.

PiyushRudani, 2016-07-11:
I have same problem as "**Sn0wDev**", the device is not entering into restore mode..
on console it is showing about to restore  then waiting for device for a while.....
and one more thing device's screen goes black (**_it is not a dfu mode_**), i can see some light is glowing.

p.s. I'm using windows 10 64bit .....

i'm atteching log. it might help.
[log.txt](https://github.com/libimobiledevice/idevicerestore/files/357211/log.txt)

nikias, 2016-07-11:
@PiyushRudani there must be something really broken. iOS 8.1.2 is not signed anymore by Apple so how would you get a response from Apple's TSS servers? Really interested in a run with -d

Sum1n, 2016-08-09:
you can use this video to fix this problem restore idevice on kali linux using libmobiledevice
https://www.youtube.com/watch?v=qRin2ggBGjw

nikias, 2016-10-04:
@PiyushRudani @Sn0wDev You never really answered my question. Please check if the device shows an Apple Logo in the middle and an (empty) progress bar below it at the point where it fails. Also @PiyushRudani I saw you ran idevicerestore with -s but this will not work unless you have an old device (iPhone 4 or below) with a bootrom exploit.
Closing for now. Feel free to create a new issue with all relevant information like Platform, device, the command line you used to run idevicerestore, and don't forget to run it with -d parameter and post the output.

closed 2016-10-04 by nikias