libimobiledevice/idevicerestore #111: Question about erasing

Peterede, 2016-07-12:
When using iDeviceRestore, are all previous bytes written over?  For example, contacts, email addresses, bookmarks, calendar events, user apps, etc. that were on the phone are no longer recoverable by *any* means?  The reason I am asking is due to privacy issues when selling a phone to another person.

Is the memory zeroed out or are the files just deleted and are able to be recovered by viewing the raw bytes on the disk?

nikias, 2016-07-12:
A restore with idevicerestore will destroy all data on disk when using the -e command line switch, otherwise the user data partition will remain untouched.
The bytes are not written over and there's no need for it. The file systems on iDevices are encrypted with a randomly created - and hardware-dependent - key. When the key is erased all data becomes unrecoverable, and this happens
a) when doing (on device) Settings -> General -> Reset -> Erase All Content and Settings
b) when doing a firmware restore with idevicerestore with -e command line switch
c) when doing a firmware restore with iTunes (Restore, not Update)

closed 2016-07-12 by FunkyM

Al22lan, 2017-05-24:
Is it possible to do the Erase All Content and Settings using the functionality of the libimobiledevice, as if it is made when you press a menu function reset of device?

yashsharma45, 2018-05-05:
@Al22lan Did you get any way to do factory Reset using LibiMobileDevice? I am also trying same but sometimes -e not works while other times it Downloads Latest Firmware. I dont want to Upgrade Firmware and want to completely erase data from device using LibiMobileDevice. How can i do that?