libimobiledevice/idevicerestore #113: Cant get idevicerestore to compile on Kali Linux

weldone, 2016-08-28:
After downloading numerous of downloading tools and other incomplete files in order to complete the installations of the idevicerestore, im now stuck at this part.
Can someone please tell me what to do or what to install ?

; root@kali:~# cd '/root/Downloads/idevicerestore-master'
root@kali:~/Downloads/idevicerestore-master# cd '/root/Downloads/idevicerestore-master/idevicerestore'
root@kali:~/Downloads/idevicerestore-master/idevicerestore# sh autogen.sh
libtoolize: putting auxiliary files in '..'.
libtoolize: linking file '../ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
configure.ac:33: installing '../compile'
configure.ac:8: installing '../missing'
src/Makefile.am: installing '../depcomp'
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
[... 125 lines cut ...]
dfu.c:271:48: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
^~~~~~~~~~~~~~~~~~~~~
dfu.c:276:21: error: dereferencing pointer to incomplete type ‘const struct irecv_device_info’
return (device_info->ibfl & IBOOT_FLAG_IMAGE4_AWARE);
^~
dfu.c: In function ‘dfu_get_ap_nonce’:
dfu.c:286:48: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
^~~~~~~~~~~~~~~~~~~~~
dfu.c:291:17: error: dereferencing pointer to incomplete type ‘const struct irecv_device_info’
if (device_info->ap_nonce && device_info->ap_nonce_size > 0) {
^~
dfu.c: In function ‘dfu_get_sep_nonce’:
dfu.c:310:48: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
const struct irecv_device_info *device_info = irecv_get_device_info(client->dfu->client);
^~~~~~~~~~~~~~~~~~~~~
dfu.c:315:17: error: dereferencing pointer to incomplete type ‘const struct irecv_device_info’
if (device_info->sep_nonce && device_info->sep_nonce_size > 0) {
^~
Makefile:624: recipe for target 'idevicerestore-dfu.o' failed
make[1]: ** [idevicerestore-dfu.o] Error 1
make[1]: Leaving directory '/root/Downloads/idevicerestore-master/idevicerestore/src'
Makefile:427: recipe for target 'install-recursive' failed
make: *** [install-recursive] Error 1

anamhira47, 2016-10-08:
I got the same error, searched the web could not find a fix for it.

slacka, 2017-01-05:
I get same error on Ubuntu 16.10
gcc (Ubuntu 6.2.0-5ubuntu12) 6.2.0 20161005

danielholm, 2017-02-15:
Same here.
gcc version 6.2.0 20161005 (Ubuntu 6.2.0-5ubuntu12)

nikias, 2017-06-15:
Make sure that you have latest libirecovery built and installed from source.

closed 2017-06-15 by nikias