libimobiledevice/idevicerestore #116: Restoring iPhone 7 Plus does not complete. ("Unknown data request 'FirmwareUpdaterData' received")

rekyagis, 2016-09-27:
When I try to restore iPhone 7 Plus using idevicerestore (from latest HEAD of git repository), it stops with the message "Unknown data request 'FirmwareUpdaterData' received".

- iPhone: iPhone 7 Plus (MNRC2J/A, A1785)
- iOS: 10.0.2
- idevicerestore, libirecovery, libimobiledevice: latest HEAD of git repository (Sep. 26th)
- OS: Ubuntu 16.04

The command line I tried is:
> idevicerestore -d -e iPhone9,4_10.0.2_14A456_Restore.ipsw 

The output message is attached:
[idevicerestore_log.txt](https://github.com/libimobiledevice/idevicerestore/files/494408/idevicerestore_log.txt)

The last message is:
>  Unknown data request 'FirmwareUpdaterData' received

Thank you for helping.

rekyagis, 2016-09-27:
I took the log what iTunes does on USB bus.

When 'FirmwareUpdaterData' is requested, iTunes responds:
>
&lt;plist version="1.0"&gt;
&lt;dict&gt;
	&lt;key&gt;FirmwareResponseData&lt;/key&gt;
	&lt;dict&gt;
		&lt;key&gt;@ServerVersion&lt;/key&gt;
		&lt;string&gt;2.1.0&lt;/string&gt;
		&lt;key&gt;FirmwareData&lt;/key&gt;
		&lt;data&gt;
		MIMOO2cCAQAwgw47XzCCBrCAAgAJgRlDMjAyMTFKRzRJSEVMX2RldlVwZGF0
... (About 910Kbytes, this is "Firmware/SE/Icefall.RELEASE.sefw". )
		AA==
		&lt;/data&gt;
		&lt;key&gt;SE,Ticket&lt;/key&gt;
		&lt;data&gt;
		MIIElxYESU00TQIBADGCAb3/hOqFnEKCAbQwggGwFgRNQU5CMYIBpv+E6oWc
... (About 1.1Kbytes, I can not find what this is. )
		iobvKDVO0sTD
		&lt;/data&gt;
	&lt;/dict&gt;
&lt;/dict&gt;
&lt;/plist&gt;

After 'FirmwareUpdaterData' is responded, another data named 'FUDData' is requested:
> 
&lt;plist version="1.0"&gt;
&lt;dict&gt;
	&lt;key&gt;DataType&lt;/key&gt;
	&lt;string&gt;FUDData&lt;/string&gt;
	&lt;key&gt;MsgType&lt;/key&gt;
	&lt;string&gt;DataRequestMsg&lt;/string&gt;
&lt;/dict&gt;
&lt;/plist&gt;

And iTunes responds for 'FUDData':
> 
&lt;plist version="1.0"&gt;
&lt;dict&gt;
	&lt;key&gt;FUDImageData&lt;/key&gt;
	&lt;dict&gt;
		&lt;key&gt;AOP&lt;/key&gt;
		&lt;data&gt;
		MIMbnt0WBElNRzQwgxuGMBYESU00UBYEYW9wZhYBMQSDG4Yc0wAAAAIAAIEJ
... (About 1.72Mbytes, I can not find what this is. )
		UroeNnd2LkrTSlW/XKnCjjK751qi
		&lt;/data&gt;
		&lt;key&gt;Homer&lt;/key&gt;
		&lt;data&gt;
		MILAQhYESU1HNDCCp5YWBElNNFAWBGhvbXIWATEEgqeDPGRpY3Q+PGtleT5D
... (About 48Kbytes, I can not find what this is. )
		Ci4qNTXoQzSP83Cb1CMSvlK6HjZ3di5K00pVv1ypwo4yu+daog==
		&lt;/data&gt;
	&lt;/dict&gt;
&lt;/dict&gt;
&lt;/plist&gt;

The USB log is attached:
[usb_monitor_log.txt.zip](https://github.com/libimobiledevice/idevicerestore/files/494613/usb_monitor_log.txt.zip)

Thanks.

nikias, 2016-09-27:
@rekyagis Thanks for reporting and the detailed info. I will see what I can do to implement this. It seems like there's another TSS request involved. You don't happen to be able to dump the communication to https://gs.apple.com using CharlesProxy or MITMproxy ?

rekyagis, 2016-09-27:
I dumped the communication between iTunes and gs.apple.com with mitmproxy.

The flow is attached:
[mitmproxy_flow.zip](https://github.com/libimobiledevice/idevicerestore/files/495452/mitmproxy_flow.zip)

We can see 'SE.Ticket' in 4th flow.
But I have still no idea for 'FUDImageData'.

Thanks.

nikias, 2016-09-27:
FUDImageData is implemented already.

nikias, 2016-09-28:
@rekyagis thanks for providing the dump. I have all the missing pieces and will update the code appropriately.

rekyagis, 2016-09-28:
@nikias I found the implementation for FUDData on the latest code. Sorry to bother you.
I am looking forward to update.
Please let me know if there is something I can do.

nikias, 2016-09-28:
@rekyagis you're not bothering me, no worries! You have been really helpful in trying to resolve that issue. I will post an update here for you to test whenever I implemented it.

nikias, 2016-09-29:
I created the following patch that *should* work. Let me know if it works.

https://www.dropbox.com/s/jucoof4if3edcp5/se_tss_request.diff?dl=1

rekyagis, 2016-09-29:
@nikias Thank you for providing the patch file.

It works with some additional updates.
The additional patch file is attached:
[se_tss_request-2.diff.zip](https://github.com/libimobiledevice/idevicerestore/files/500670/se_tss_request-2.diff.zip)
(NOTE: This patch should be applied after se_tss_request.diff on above comment.)
Could you review it.

Thanks.

nikias, 2016-09-29:
Looks good! I made some more logical and semantic changes. Please try this new patch (revert previous patches!). Thanks for your help.
https://www.dropbox.com/s/lzm3kl28xmwqdi4/se_tss_request_new.diff?dl=1

rekyagis, 2016-09-30:
It works fine!
Thank you for your great work.

rekyagis, 2016-09-30:
This is not a functional issue, but I have found one thing.
A message "ERROR: No 'SE,Ticket' in TSS response, this might not work" (at restore.c:1794) is shown in normal case. The condition for it is opposite.
Thanks.

pasike, 2016-09-30:
We tested with the patch and iPhone 9,1. Works fine, thank you!

nikias, 2016-09-30:
Thanks! Will fix the condition for the error message and push the commit.

nikias, 2016-09-30:
@pasike @rekyagis sorry, I rewrote the code a bit (again) to match other TSS code that we already have. These are the final patches (if everything works as expected).
Would you mind testing them once more before I commit them to master?
Thanks!

https://www.dropbox.com/s/80d5k7dwnd2o9cm/0001-tss-Add-helper-function-to-create-Secure-Element-TSS.patch?dl=1
https://www.dropbox.com/s/sgxiwmj19e9vzlq/0002-restore-Add-support-for-FirmwareUpdaterData-request-.patch?dl=1

pasike, 2016-10-02:
Will test on Monday.

rekyagis, 2016-10-03:
It works fine with my iPhone 7 Plus.

The output messages are attached:
[idevicerestore_log.txt](https://github.com/libimobiledevice/idevicerestore/files/505400/idevicerestore_log.txt)
[idevicerestore-d_log.txt](https://github.com/libimobiledevice/idevicerestore/files/505401/idevicerestore-d_log.txt)

Thanks.

pasike, 2016-10-03:
It works with iPhone 7 (9,1). Thanks!

closed 2016-10-03 by nikias