libimobiledevice/idevicerestore #118: Restore failed in recovery mode on iPhone 7 (iphone9,3)

truonggiang0710, 2016-10-03:
Hi guys,

I used last version to restore iPhone 7 in recovery mode. But the restored is run forever.
The error below. Please help to to resole the issue.

FDR 0xb3b62e80 waiting for message...
FDR 0xb3b62e80 waiting for message...
FDR 0xb3a02548 waiting for message...
FDR 0xb3a02548 terminating...
FDR 0xb3b62e80 waiting for message...
FDR 0xb3a02860 waiting for message...
FDR 0xb3b62e80 waiting for message...
FDR 0xb3a036f8 waiting for message...
FDR 0xb3a036f8 terminating...
FDR 0xb3b62e80 waiting for message...
FDR 0xb3a02860 terminating...
FDR 0xb3b62e80 waiting for message...
FDR 0xb3b62e80 waiting for message...
FDR 0xb3b62e80 waiting for message...
FDR 0xb3b62e80 waiting for message...

nikias, 2016-10-03:
Can you run it again with -d and upload the output?

nikias, 2016-10-18:
@truonggiang0710 does it work now? what about a debug run with -d as I requested?

truonggiang0710, 2016-10-19:
Thanks you @nikias .
I ran again with other network and it's OK. I think slowly of my network is main reason.
Thanks a lot.

closed 2016-10-19 by truonggiang0710